SpinalHDL/SpinalHDL #317: Generate C header file from Bus Slave Factory

c-thaler, 2020-05-08:
It would be nice to generate a C header file containing neat defines for the register offsets. These defines could be used in a C driver to address the registers.
Currently, `BusSlaveFactory` already has a documentation string for each register. For the C header there is also a need for a brief register name. A parameter `name` would have to be added to the `BusSlaveFactory` functions besides the `documentation` string.
Internally, there may have to be distinguished between register names for read and write access. In some cases, read and write to the same offset has different functionality/effects.

It would be great to have have an additional generator for JSON/XML/whatever. The output can be used used for documentation purpose.

jijingg, 2020-05-09:
@c-thaler yes, i agree, its usefull to generator C head,
currently,`busSlaveFctory`/`busif` could generate register interface HTML documents (#238 have not merged yet)
[image]
so it easy to dump head file like this,
```c
#define GNSS_AE_ADDRESS_BASE
#define R_CLOCK         = GNSS_AE_ADDRESS_BASE  +  0x00
#define R_SWEEP_CONFIG  = GNSS_AE_ADDRESS_BASE  +  0x04
#define R_CACQ_CONFIG   = GNSS_AE_ADDRESS_BASE  +  0x08
...
#define R_INT_STATUS    = GNSS_AE_ADDRESS_BASE  +  0x80
```
but i'm not sure is it neccssary to generator field struct of REG something like following code,  i not very familiar with C, so if you have example, templete  or  advice,  that would be better
```c
typedef union{
    uint32_t val;
    struct{
        uint32_t sweep_range:5;
        uint32_t aiding_range:3;
        uint32_t reserved:24;
    } reg;
} nSWEEP_CONFIG
```

gyn, 2020-05-09:
> @c-thaler yes, i agree, its usefull to generator C head,
> currently,`busSlaveFctory`/`busif` could generate register interface HTML documents (#238 have not merged yet)
> [image]
> so it easy to dump head file like this,
> 
> ```c
> #define GNSS_AE_ADDRESS_BASE = 0x1F00_2000
> 
> #define R_CLOCK         = GNSS_AE_ADDRESS_BASE  +  0x00
> #define R_SWEEP_CONFIG  = GNSS_AE_ADDRESS_BASE  +  0x04
> #define R_CACQ_CONFIG   = GNSS_AE_ADDRESS_BASE  +  0x08
> ...
> #define R_INT_STATUS    = GNSS_AE_ADDRESS_BASE  +  0x80
> ```
> 
> but i'm not sure is it neccssary to generator field struct of REG something like following code, i not very familiar with C, so if you have example, templete or advice, that would be better
> 
> ```c
> typedef union{
>     uint32_t val;
>     struct{
>         uint32_t sweep_range:5;
>         uint32_t aiding_range:3;
>         uint32_t reserved:24;
>     } reg;
> } nSWEEP_CONFIG
> ```
If bit-field code was produced, you should think about endianness. So the code is more like:

typedef union {
    uint32_t val;
    struct{
#if defined(LITTLE_ENDIAN)
        uint32_t sweep_range:5;
        uint32_t aiding_range:3;
        uint32_t reserved:24;
#elif defined(BIG_ENDIAN)
        uint32_t reserved:24;
        uint32_t aiding_range:3;
        uint32_t sweep_range:5;
#else
#error "Unknown byte order"
#endif
    } reg;
} nSWEEP_CONFIG;

jijingg, 2020-05-10:
@gyn  , emm, the endianness should be considered on RAM, but the bit wise have no concept of endianeess, it's only for byte, right ? because the field doesn't have to be byte, it may be 1, 3 5bit  or any other widths. so the c struct must one by one match the field defined in regif.

c-thaler, 2020-05-10:
@jijingg Yes, you're right. Endianness is about byte order, not bit order. But the bit order is still a problem, because it is implementation dependent. It is unfortunately not defined in the C standard.

Anyway, that's great work you proposed here and it is exactly what I have been looking for.

It would be nice to add a "_t" to end of the type name and have it lower case. That is what you usually find in the Linux kernel.
It would also be great to add a user defined prefix to the type name. In the example below I took "gnss" from your register documentation as a prefix. 

```
typedef union{
    uint32_t val;
    struct{
        uint32_t sweep_range:5;
        uint32_t aiding_range:3;
        uint32_t reserved:24;
    } reg;
} gnss_sweep_config_t
```

In the header file, you can definitely skip the base address define `GNSS_AE_ADDRESS_BASE`. Base address is determined during system integration and driver writers should handle the base address in their driver's "init" function. In some systems you may even have multiple instances of the same IP core.
Defines with the byte offset are sufficient. Again it would be great to add a user defined prefix to the defines like in the example below.

```
#define GNSS_R_CLOCK         = 0x00
#define GNSS_R_SWEEP_CONFIG  = 0x04
```

jijingg, 2020-05-10:
@c-thaler right, the prefix is necessary . and the `XXXX_ADDRESS_BASE` should not be hard coded.

> In some systems you may even have multiple instances of the same IP core... Defines with the byte offset are sufficient.

yes ,it's make sense.

jijingg, 2020-05-10:
@c-thaler  the work is done , #238 have updated , you can fork my regif branch , then `sbt clean publishLocal` try that.

example :https://github.com/jijingg/SpinalHDL/blob/regif/tester/src/main/scala/spinal/tester/code/RegIfExample.scala#L202
will generator 3 file 
```shell
RegIfExample.h
RegIfExample.html
RegIfExample.v
```
RegIfExample.h: 
```c
#define APM_TURBO_EARLY_QUIT    = 0x0
#define APM_TURBO_THETA         = 0x4
#define APM_TURBO_START         = 0x8
#define APM_TURBO_MOD           = 0xc
#define APM_TURBO_CRC_POLY      = 0x10
#define APM_TURBO_K             = 0x14
#define APM_TURBO_F1F2          = 0x18
#define APM_TURBO_MAX_ITER      = 0x1c
#define APM_TURBO_FILL_NUM      = 0x20
#define APM_TURBO_3G_INTER_PV   = 0x24
#define APM_TURBO_3G_INTER_CRP  = 0x28
#define APM_TURBO_3G_INTER_FILL = 0x2c

typedef union{
    uint32_t val;
    struct{
        uint32_t early_quit:1;
        uint32_t early_count:2;
        uint32_t na_:29;
    } reg;
} apm_turbo_early_quit_t

typedef union{
    uint32_t val;
    struct{
        uint32_t theta:4;
        uint32_t bib_order:1;
        uint32_t na_:27;
    } reg;
} apm_turbo_theta_t                             
```

gyn, 2020-05-10:
> @gyn , emm, the endianness should be considered on RAM, but the bit wise have no concept of endianeess, it's only for byte, right ? because the field doesn't have to be byte, it may be 1, 3 5bit or any other widths. so the c struct must one by one match the field defined in regif.

You should also consider endianness when using bit-filed, There is also other kinda endianess. :) Check https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/include/uapi/linux/ip.h?h=v5.7-rc4#n86 if you have doubt.

c-thaler, 2020-05-11:
@jijingg I've just tried it. Works like a charm :+1:

andreasWallner, 2020-05-12:
@jijingg Sorry to butt in so late...
It would be great if one could still generate other ouputs from the model.
I'd like to e.g. generate header files that look like the CMSIS headers, information to work with [kvasir-io](https://github.com/kvasir-io/Kvasir), and IP-XACT outputs.
I had a short look on the code but did not have enough time to check if that is already possible... (and I'd not suggest to hold off on merging this if not already possible)

jijingg, 2020-05-12:
@andreasWallner what is CMSIS headers, could you give me a example of CMSIS/IP-XACT .
Generally, any format can be generated as long as you provide a template。
the current branch have merged. so we can added next version.  for more detail, refer to #237

andreasWallner, 2020-05-13:
@jijingg 

Example for CMSIS Headers would be (for your first example in #237, not complete):

    typedef struct {
      __IO uint32_t REG0;
      __IO uint32_t REG1;
      __IO uint32_t REG2;
          uint32_t 
      __IO uint32_t REGn;
      __IO uint32_t REGn1;
    } RegBankExample_Type;

    #define REG_BANK_EXAMPLE_BASE 0xDEADAFFE
    #define REG_BANK_EXAMPLE ((RegBankExample_Type*)REG_BANK_EXAMPLE_BASE)

    #define REG_IF2_REG0_FIELDS_0_Pos 0
    #define REG_IF2_REG0_FIELDS_0_Msk UINT32_C(0x3)
    // skipping the others...
    #define REG_IF2_REG0_FIELDS_3_Pos 16
    #define REG_IF2_REG0_FIELDS_3_Msk UINT32_C(0xF0000)

These are currently pretty much universally delivered by processor vendors for interacting with their peripherals. (sometimes split across files, differences in the detailed naming, but I guess you get the idea)

IP-XACT is an XML format used by many tools as an exchange format for (among other things) registers. 

But you might have misunderstood, it was not a request for you to implement a specific format, more so the question if it would be possible to build the code in a way that one can easily generate different/additional outputs. E.g. by registering a generator before running HW generation.

I'd be more than willing to contribute specific generators like for the format shown above. I just see the need to be able to generate outputs where there might not be a generator in the SpinalHDL code base (in my case e.g. I'd want to generate proprietary files for our internal tools which won't make much sense for anyone else, e.g. because they are just there to integrate with legacy tools)

jijingg, 2020-05-13:
@andreasWallner  the current HTML document and C Head complete automatic,  but we can close it in SpinalConfig( update in next version), and it also easy to add your custom format, and dump manully at the bottom of file 
```scala 
class MyRegIF extends  Component {
  val io = new Bundle{
    val ahb = AhbLite3(AhbLite3Config(8, 32))
  }

  val busif = BusInterface(io.ahb, (0x00, 1 KiB))

  val reg0 = busif.newReg(doc = "xxxx")
  val reg1 = busif.newReg(doc = "xxxx")
  val fd0 = reg1.field(2 bits, AccessType.RW, 0, doc = "field 0")
  val fd1 = reg1.field(8 bits, AccessType.RW, 0, doc = "field 0")
  val reg2 = busif.newReg(doc = "xxxx")

 
  busif.document("docName")    //currently you do not need manully dot this , it complete automatic
  busif.genCHead("cHeadName") 

  // but you custom Fromat  you can manully dump it 
  busif.GenMyFormat("yourName") 
}
```
the custom Format not necessarily Modify the source code(if it really common , welcome add them to regif dir),  if the format is  very personnal ,  just add `GenMyFormat` by implicit extends like following, then you can access all data struct which stored in x.RegInsts( currently it's  private, i will add getRegInsts method reopen that, so you can get the full data stuct)
```scala 
object YourBusIfExends {
  implicit class BusIfExends(x: BusIf) {
    def GenMyFormat() = {
      x.getRegInsts then do what ever you want do
    }
  }
}
```
ref :  
[cHead](https://github.com/SpinalHDL/SpinalHDL/blob/dev/lib/src/main/scala/spinal/lib/bus/regif/BusIfBase.scala#L134)
 [CHead format ](https://github.com/SpinalHDL/SpinalHDL/blob/c1f40bcec9c055150a1789fce1a6628ed073244e/lib/src/main/scala/spinal/lib/bus/regif/Document/CHeads.scala#L26)

c-thaler, 2020-05-14:
Some kind of visitor pattern may help.
@jijingg You can implement only the handling of the visitor and keep your internal data structures private.
@andreasWallner You get simple callbacks with some parameters following the internal BusIf structure.

I did a quick and dirty implementation that you can try out to get the idea:
https://github.com/c-thaler/SpinalHDL/blob/25ed77669999bc4b73cf0dcfc3b6783924ea2033/tester/src/main/scala/spinal/tester/code/RegIfExample.scala#L202

To add a custom generator, one has to implement the BusIfVisitor trait:
https://github.com/c-thaler/SpinalHDL/blob/25ed77669999bc4b73cf0dcfc3b6783924ea2033/lib/src/main/scala/spinal/lib/bus/regif/BusIfBase.scala#L12

jijingg, 2020-05-14:
@c-thaler  i'm ready to add a method of `def getRegInsts() = RegInsts.toList`, so the `RegInsts` still  safe as private. But provide a visitor is good idea, more elegant.

c-thaler, 2020-05-14:
@jijingg I can go on with implementation if you want.

jijingg, 2020-05-14:
@c-thaler go on, I appreciate it ^^

c-thaler, 2020-05-14:
Almost finished with the C header generator:

```
/*
 * Bus Interface Header - AUTOGENERATED
 */

#ifndef AP_REGIF_H
#define AP_REGIF_H

#define AP_M_TURBO_EARLY_QUIT    0x00
#define AP_M_TURBO_THETA         0x04
#define AP_M_TURBO_START         0x08
#define AP_M_TURBO_MOD           0x0c
#define AP_M_TURBO_CRC_POLY      0x10
#define AP_M_TURBO_K             0x14
#define AP_M_TURBO_F1F2          0x18
#define AP_M_TURBO_MAX_ITER      0x1c
#define AP_M_TURBO_FILL_NUM      0x20
#define AP_M_TURBO_3G_INTER_PV   0x24
#define AP_M_TURBO_3G_INTER_CRP  0x28
#define AP_M_TURBO_3G_INTER_FILL 0x2c

typedef union {
	u32 val;
	struct {
		u32 fillPos     :  9;
		u32 reserved_0  :  7;
		u32 fillRow     :  2;
		u32 reserved_1  : 14;
	} reg;
} ap_m_turbo_early_quit_t;
```

jijingg, 2020-05-17:
@c-thaler good, with a better looking

numero-744, 2022-10-03:
I think this issue can be closed as completed

closed 2022-10-04 by Dolu1990