SpinalHDL/SpinalHDL #319: Can I use localparameter in FSM/Enum instead of define?

lifoo, 2020-05-14:
Question1：
Why is the state of FSM and Emun defined with define? This introduces hidden risks, for example:
There are 3 verilog files, a.v, b.v and c.v. The synthesizer reads the files in sequence, a.v is my handwritten code with verilog, there is a define DEFINE_USE_IN_A_C, b.v is a verilog file generated by SpinalHDL, there is a state machine or Enum, it happens to redefine DEFINE_USE_IN_A_C, but value is not equal as in a.v, c.v is also handwritten code,did not define DEFINE_USE_IN_A_C, but used DEFINE_USE_IN_A_C, but DEFINE_USE_IN_A_C has been redefined in b.v, resulting in cv circuit error.
I think it would be better use localparameter to solve the problem in verilog.

Question2：
`ifndef SYNTHESIS
...
`endif
When using FSM or Enum, can the above codes be generated selectively? Otherwise, I need to remind back-end engineers to define SYNTHESIS to continue their work.

Dolu1990, 2020-05-14:
Hi,
> Question1：

Hoo, so in Verilog, define definitions can cross from one .v to another ?
or you mean when you include a verilog file into another ?
(sorry, i was raised as a VHDL guy, don't know so much about Verilog :))

> Question2：

Sorry i'm not sure to understand the exact case you ask. You mean that you want to have some generated code to be in ifndef SYNTHESIS ... endif ? or you ask about some ifndef SYNTHESIS ... endif which are generated by SpinalHDL itself already ?

lifoo, 2020-05-15:
Hi Dolu:
Question1:
Verilog's macro definition is global from the definition point, and remains valid for subsequent files (this is different from C language) until another macro definition changes the value of this macro definition, or use undef to cancel this Macro definition. Since the macro definition remains valid for the files that are subsequently read, it is usually required to compile the files in a certain order.
The most typical problem with using macro definitions is that another file's macro definition uses the same name. When encountering this kind of problem, it may take a lot of cost to debug.
Question2:
When using StateMachine or Enum, the following verilog code will be generated:
`ifndef SYNTHESIS
  always @(*) begin
    case(fsm_stateReg)
      `fsm_enumDefinition_binary_sequential_boot : fsm_stateReg_string = "boot      ";
      `fsm_enumDefinition_binary_sequential_fsm_stateA : fsm_stateReg_string = "fsm_stateA";
      `fsm_enumDefinition_binary_sequential_fsm_stateB : fsm_stateReg_string = "fsm_stateB";
      `fsm_enumDefinition_binary_sequential_fsm_stateC : fsm_stateReg_string = "fsm_stateC";
      default : fsm_stateReg_string = "??????????";
    endcase
  end
`endif
This code will not be used during synthesis, I guess the original intention of adding this code is for debugging convenience. But verilog's parameter / localparameter can also achieve the same effect. If I use localparameter, the localparameter string will be displayed instead of value in the waveform (i use synopsys vcs),and the method of using localparameter is synthesizable.
I need to define macro SYNTHESYS when synthesizing, otherwise synthesis will report an error, and design and synthesis are done by different people in my company, which increases the cost of communication.So can the generation of the above code be controlled?Like"In order to avoid the prefix “io_” on each of the IOs of the blackbox, you can use the function noIoPrefix()"

Dolu1990, 2020-05-19:
Hi @lifoo 

> Question1:
Ahhh what's a paine in the ass ^^.
I just checked, localparam can't be used globaly in a file right ? Or how would you do it ?

> Question2:
Hoo right, the debug strings. So i have to play around with localparameter to figure out things :)
Hmm else ifdef SIMULATION   instead of ifndef SYNTHESIS
Any way, yes, you are right, this need to be fix one way or another.

Regards
Charles

lifoo, 2020-05-22:
Hi @Dolu1990 
Sorry, this week's work is too busy, so I reply to you so late, thank you very much for listening to my suggestions.

I wrote a simple example to help you understand the common writing of FSM in verilog：
```
module fsm_example(
    input       clk,
    input       rst_n,
    input       i_data, 
    output reg  o_result
);
 
//use localparam as state define
//we can use parameter too,but it's better use localparam in FSM state
//If there is a parameter in the submodule, 
//the parent module can pass parameters to the submodule when it instantiates the submodule. 
//However, the FSM state machine is only used by the module itself, 
//so it is safer to use localparam.
localparam IDLE = 2'b00;
localparam S0   = 2'b01;
localparam S1   = 2'b10;

reg[1:0] current_state;
reg[1:0] next_state;
//current state
always @ (posedge clk or negedge rst_n) begin
    if(!rst_n)
        current_state <= IDLE;
    else
        current_state <= next_state;
end
//next state
//just detect 'b111 sequence
always @ (*) begin
    case(current_state)
        IDLE:begin
            if(i_data) next_state = S0;
            else next_state = IDLE;
        end
        S0:begin
            if(i_data) next_state = S1;
            else next_state = IDLE;
        end
        S1:begin
            if(i_data) next_state = S1;
            else next_state = IDLE;
        end
        default : next_state = IDLE;
   endcase
end
//output logic
always @ (posedge clk or negedge rst_n) beign
    if(!rst_n)
        o_result <= 1'b0;
    else
        o_result <= current_state == S1;
end

endmodule
```

hudajiang, 2020-06-12:
Localparam of FSM state is nice to me，too。it is very friendly for debuy in Verdi

Dolu1990, 2020-06-13:
Sorry, completly forgot to track it.

One question i have is  : 

If somewere in the hearchy i set  signal X with a local param (as a enum), then carry that signal X to somewhere else in the hearchy to the signal Y, does that signal Y keep the name of local params used for signal X ? or it lose it and become a single binary value in the wave ?

In otherwords, does local param works great for enum which cross hearchy ?

hudajiang, 2020-06-14:
1. Local param can only use in the verilog module who define it .  like local var , private var... 
2.  Some EAD tool (Like Verdi VCS) can extract the FSM from verilog code.  If  use local param (or param) for state of FSM ,after EAD tool   extracted  it,  we can see the STATE name of param in the EAD tool everywhere instead of binary value. Nice to debug
3.eg. we can see the the STATE name of param instead of  binary value in wave 

Forgive me for my poor english XD

saahm, 2020-06-14:
The two latter points you mention bring some general property up: Usually Verilog does not know anything else than its basic datatypes: reg and wire (putting aside some specialties, i want to draw the line between vhdl and verilog here). In VHDL you can have an actual Enum type like in C or other languages. Everytime you do a construction that behaves like an enum in Verilog, this is basically just constants that you name and encode yourself. 
Using localparam is conventionally the way to go with the scoping (because param can be read from outside in simulations, but for synthesis this all gets dropped). 
The usual way to go without additional third party tools, when you want to debug an FSM you either stay with the encoding you gave yourself or you apply addtional constructs in your design that arent synthesizable. See [this](https://github.com/cliffordwolf/picorv32/blob/f9b1beb4cfd6b382157b54bc8f38c61d5ae7d785/picorv32.v#L1181) example from Clifford Wolfs picorv32 implementation. In the lines above he defines localparams, then he adds his ascii-representations from his FSM states. Otherwise he'd have to look for "8'b10000000" in his simulation for the state.
I think the idea between param and localparam is: 
* Use param for generics like in VHDL
* Use localparam for state declaration, local constants (could also use #define but these get treated differently i think)

In [this](https://stackoverflow.com/questions/30288783/difference-between-parameter-and-localparam/30288884#30288884) stackoverflow discussion the point being is: localparams cannot be changed from outside while param can be changed from outside.

wswslzp, 2020-10-15:
Is there any progress for this state machine issue?

Dolu1990, 2020-10-15:
Hi
No progress was done actualy.

tomverbeure, 2021-01-04:
> But verilog's parameter / localparameter can also achieve the same effect. If I use localparameter, the localparameter string will be displayed instead of value in the waveform (i use synopsys vcs),and the method of using localparameter is synthesizable.

I would also like to add my own 2 cents here: commercial tools liks VCS are able to display the name of a `localparam` in their waveform viewer, but open source tools do not. That's why I asked Charles to emit text strings for enums. It would IMO be a major loss if that feature went away.

volatile-static, 2021-01-25:
Oh, sadly to find this issue has no progress.
As a student who learned Verilog and Quartus only, I really need `localparam` instead of `define`.
Thanks for your consideration ;-)

Dolu1990, 2021-01-25:
@volatile-static What is the issue with quartus ?

volatile-static, 2021-01-26:
Hi， @Dolu1990 
The previous discussion already contains my problem.
Shortly, I cannot see state names in schematic if use `define`. So, use `localparam` can help me debug.
That's all.

lifoo, 2022-11-22:
localparam has been support now，thanks guys~

closed 2022-11-22 by lifoo