SpinalHDL/SpinalHDL #330: Why there is boot state before the state with EntryPoint?

Ncerzzk, 2020-06-18:
for example ,I write the Spinahdl code below:
```scala
class MyTop2 extends Component{
  val io = new Bundle{
    val input = in Bool
    val result = out Bool
  }

  val fsm = new StateMachine{
    io.result:=False
    val stateB : State = new State with EntryPoint{
      whenIsActive {
        when(!io.input){
          goto(stateA)
        }
        io.result := True
      }
    }
    val stateA : State = new State{
      whenIsActive{
        when(!io.input)(goto(stateB))
        io.result := False
      }
    }
  }
}
```

I want to implement the state machine:
[image]

while the verilog code is not what  I want, because there is a boot state before stateB:
```verilog
  always @ (posedge clk or posedge areset) begin
    if (areset) begin
      fsm_stateReg <= `fsm_enumDefinition_binary_sequential_boot;
    end else begin
      fsm_stateReg <= fsm_stateNext;
    end
  end
```
and after one cycle, It came to StateB:
```verilog
  always @ (*) begin
    fsm_stateNext = fsm_stateReg;
    case(fsm_stateReg)
      `fsm_enumDefinition_binary_sequential_fsm_stateB : begin
        if((! inxxx))begin
          fsm_stateNext = `fsm_enumDefinition_binary_sequential_fsm_stateA;
        end
      end
      `fsm_enumDefinition_binary_sequential_fsm_stateA : begin
        if((! inxxx))begin
          fsm_stateNext = `fsm_enumDefinition_binary_sequential_fsm_stateB;
        end
      end
      default : begin
        fsm_stateNext = `fsm_enumDefinition_binary_sequential_fsm_stateB;
      end
    endcase
  end
```

So I want to ask that is there some mistakes I made? How to implement the State Machine?

Dolu1990, 2020-08-27:
Hi,

Sorry for the delay, completly forgot it :/

So, basicaly, this boot state mainly allow to "cleanly" entry the first state. It is also used to manage inner state machines. All is good from your side :)

Ncerzzk, 2020-08-28:
Thank you for your reply, I got it.

closed 2020-08-28 by Ncerzzk