SpinalHDL/SpinalHDL #331: If a module contain a state machine, it can't be replaced by the same module in the generated HDL codes.

tengleness, 2020-06-19:
i create a sub module with a state machine like this:
```class test001 extends Component {
  val io = new Bundle {
    val a = in Bool()
    val b = out Bool()
  }

  val x = Reg(Bool()) init False
  io.b := x
  val fsm = new StateMachine {
    val idle = new State with EntryPoint
    val s1 = new State()

    idle
      .whenIsActive {
        goto(s1)
      }
    s1
      .whenIsActive {
        x := io.a
        goto(idle)
      }
  }

}
```
and i instanced this module two times like this:
```class testTop extends Component {
  val io = new Bundle {
    val a = in Bool()
    val b = out Bool()
  }

  val t0, t1 = new test001
  io.a <> t0.io.a
  io.a <> t1.io.a
  io.b <> (t0.io.b ^ t1.io.b)

}
```
in the generated verilog codes, there are two test001 module: "test001" and "test001_1_"
the "test001_1_" can't be replaced by "test001" because there defines of enumDefinition are different, it result the HDL file very huge.
if the sub module hasn't state machine, the others can be replaced by the first correctly

lifoo, 2020-07-04:
Here is the generated verilog,VHDL has the same problem。

# verilog
```
// Generator : SpinalHDL v1.4.2    git head : 9645346cd7863293ab381e41cd7f39ca0c3c3591
// Component : testTop

`define fsm_enumDefinition_binary_sequential_type [1:0]
`define fsm_enumDefinition_binary_sequential_boot 2'b00
`define fsm_enumDefinition_binary_sequential_fsm_idle 2'b01
`define fsm_enumDefinition_binary_sequential_fsm_s1 2'b10

`define fsm_enumDefinition_1_binary_sequential_type [1:0]
`define fsm_enumDefinition_1_binary_sequential_boot 2'b00
`define fsm_enumDefinition_1_binary_sequential_fsm_idle 2'b01
`define fsm_enumDefinition_1_binary_sequential_fsm_s1 2'b10

module testTop (
  input               io_a,
  output              io_b,
  input               clk,
  input               reset
);
  wire                t0_io_b;
  wire                t1_io_b;

  test001 t0 (
    .io_a     (io_a     ), //i
    .io_b     (t0_io_b  ), //o
    .clk      (clk      ), //i
    .reset    (reset    )  //i
  );
  test001_1 t1 (
    .io_a     (io_a     ), //i
    .io_b     (t1_io_b  ), //o
    .clk      (clk      ), //i
    .reset    (reset    )  //i
  );
  assign io_b = (t0_io_b ^ t1_io_b);

endmodule

module test001_1 (
  input               io_a,
  output              io_b,
  input               clk,
  input               reset
);
  reg                 x;
  wire                fsm_wantExit;
  reg        `fsm_enumDefinition_1_binary_sequential_type fsm_stateReg;
  reg        `fsm_enumDefinition_1_binary_sequential_type fsm_stateNext;
  `ifndef SYNTHESIS
  reg [63:0] fsm_stateReg_string;
  reg [63:0] fsm_stateNext_string;
  `endif

  `ifndef SYNTHESIS
  always @(*) begin
    case(fsm_stateReg)
      `fsm_enumDefinition_1_binary_sequential_boot : fsm_stateReg_string = "boot    ";
      `fsm_enumDefinition_1_binary_sequential_fsm_idle : fsm_stateReg_string = "fsm_idle";
      `fsm_enumDefinition_1_binary_sequential_fsm_s1 : fsm_stateReg_string = "fsm_s1  ";
      default : fsm_stateReg_string = "????????";
    endcase
  end
  always @(*) begin
    case(fsm_stateNext)
      `fsm_enumDefinition_1_binary_sequential_boot : fsm_stateNext_string = "boot    ";
      `fsm_enumDefinition_1_binary_sequential_fsm_idle : fsm_stateNext_string = "fsm_idle";
      `fsm_enumDefinition_1_binary_sequential_fsm_s1 : fsm_stateNext_string = "fsm_s1  ";
      default : fsm_stateNext_string = "????????";
    endcase
  end
  `endif

  assign io_b = x;
  assign fsm_wantExit = 1'b0;
  always @ (*) begin
    fsm_stateNext = fsm_stateReg;
    case(fsm_stateReg)
      `fsm_enumDefinition_1_binary_sequential_fsm_idle : begin
        fsm_stateNext = `fsm_enumDefinition_1_binary_sequential_fsm_s1;
      end
      `fsm_enumDefinition_1_binary_sequential_fsm_s1 : begin
        fsm_stateNext = `fsm_enumDefinition_1_binary_sequential_fsm_idle;
      end
      default : begin
        fsm_stateNext = `fsm_enumDefinition_1_binary_sequential_fsm_idle;
      end
    endcase
  end

  always @ (posedge clk or posedge reset) begin
    if (reset) begin
      x <= 1'b0;
      fsm_stateReg <= `fsm_enumDefinition_1_binary_sequential_boot;
    end else begin
      fsm_stateReg <= fsm_stateNext;
      case(fsm_stateReg)
        `fsm_enumDefinition_1_binary_sequential_fsm_idle : begin
        end
        `fsm_enumDefinition_1_binary_sequential_fsm_s1 : begin
          x <= io_a;
        end
        default : begin
        end
      endcase
    end
  end

endmodule

module test001 (
  input               io_a,
  output              io_b,
  input               clk,
  input               reset
);
  reg                 x;
  wire                fsm_wantExit;
  reg        `fsm_enumDefinition_binary_sequential_type fsm_stateReg;
  reg        `fsm_enumDefinition_binary_sequential_type fsm_stateNext;
  `ifndef SYNTHESIS
  reg [63:0] fsm_stateReg_string;
  reg [63:0] fsm_stateNext_string;
  `endif

  `ifndef SYNTHESIS
  always @(*) begin
    case(fsm_stateReg)
      `fsm_enumDefinition_binary_sequential_boot : fsm_stateReg_string = "boot    ";
      `fsm_enumDefinition_binary_sequential_fsm_idle : fsm_stateReg_string = "fsm_idle";
      `fsm_enumDefinition_binary_sequential_fsm_s1 : fsm_stateReg_string = "fsm_s1  ";
      default : fsm_stateReg_string = "????????";
    endcase
  end
  always @(*) begin
    case(fsm_stateNext)
      `fsm_enumDefinition_binary_sequential_boot : fsm_stateNext_string = "boot    ";
      `fsm_enumDefinition_binary_sequential_fsm_idle : fsm_stateNext_string = "fsm_idle";
      `fsm_enumDefinition_binary_sequential_fsm_s1 : fsm_stateNext_string = "fsm_s1  ";
      default : fsm_stateNext_string = "????????";
    endcase
  end
  `endif

  assign io_b = x;
  assign fsm_wantExit = 1'b0;
  always @ (*) begin
    fsm_stateNext = fsm_stateReg;
    case(fsm_stateReg)
      `fsm_enumDefinition_binary_sequential_fsm_idle : begin
        fsm_stateNext = `fsm_enumDefinition_binary_sequential_fsm_s1;
      end
      `fsm_enumDefinition_binary_sequential_fsm_s1 : begin
        fsm_stateNext = `fsm_enumDefinition_binary_sequential_fsm_idle;
      end
      default : begin
        fsm_stateNext = `fsm_enumDefinition_binary_sequential_fsm_idle;
      end
    endcase
  end

  always @ (posedge clk or posedge reset) begin
    if (reset) begin
      x <= 1'b0;
      fsm_stateReg <= `fsm_enumDefinition_binary_sequential_boot;
    end else begin
      fsm_stateReg <= fsm_stateNext;
      case(fsm_stateReg)
        `fsm_enumDefinition_binary_sequential_fsm_idle : begin
        end
        `fsm_enumDefinition_binary_sequential_fsm_s1 : begin
          x <= io_a;
        end
        default : begin
        end
      endcase
    end
  end

endmodule
```

numero-744, 2022-10-23:
This bug is still there.

Executable example:

```scala
object TestTestTop extends App {
  import spinal.core._
  import spinal.lib.fsm._

  class test001 extends Component {
    val io = new Bundle {
      val a = in Bool ()
      val b = out Bool ()
    }

    val x = Reg(Bool) init False

    val fsm = new StateMachine {
      val idle = new State with EntryPoint
      val s1 = new State

      idle.whenIsActive { goto(s1) }
      s1.whenIsActive { x := io.a; goto(idle) }
    }

    io.b := x
  }

  class testTop extends Component {
    val io = new Bundle {
      val a = in Bool ()
      val b = out Bool ()
    }

    val t0, t1 = new test001
    io.a <> t0.io.a
    io.a <> t1.io.a
    io.b <> (t0.io.b ^ t1.io.b)
  }

  SpinalSystemVerilog(new testTop)
}
```

numero-744, 2022-11-02:
@Dolu1990 I think this should be fixed ASAP. Any idea?

Dolu1990, 2022-11-02:
It isn't realy a bug, but a feature (a bad one ^^)
Each time you define a SpinalEnum instance it create a new enum definition. And currently SpinalEnum a handled as things in the global name space (not component name space)
I would say, the missing piece is some SpinalHDL phase to merge similar SpinalEnum instances together. Or to manage them localy to the component they are used, but it goes against how the VHDL backend is designed.

numero-744, 2022-11-03:
Ahh, you mean there is a mechanism / phase to notice that two `new MyComponent` are equivalent when elaborating, and here it fails because of a `SpinalEnum` built in the constructor?

Dolu1990, 2022-11-03:
Yes, each SpinalEnum instance is conssidered as unique and global => unique names

numero-744, 2022-11-05:
@Dolu1990 How hard would it be to fix it?

I mean we could fix it to have a cleaner HDL -> RTL matching, but I am wondering: how bad is it? It seems not harmful for the for the simulator, for linters, for synthesizers, for testers nor for designers. So if it is not too hard to fix, why not fixing it; else I think it is not really a problem.

Dolu1990, 2022-11-06:
I would say, it isn't necessarly too hard, it mainly require to check equivalence between Enum definition, and then, remplace all usage of the one in the netlist to replace it with the other one.

It would be fixable without too much pain i think.