SpinalHDL/SpinalHDL #342: Reuse the previously compiled code in spinal simulation

wswslzp, 2020-09-07:
When using spinal for simulation, verilator compile the rtl component every time I run the simulation but the rtl design has no change. If the rtl component part is too large the compile time will account for most of the runtime. So reuse the previously compiled code could reduce runtime. This approach has been frequently utilized by other commercial simulation tools. 

--- 

Is there any way to skip the compilation stage?

Dolu1990, 2020-09-07:
Currently there is no way to reuse compiled C++ model from previous JVM executions.
But there is a few things which can help reducing time : 
- Recent SpinalHDL versions compile verilator with multithreading, this realy help
- Not enabling "SimConfig.allOptimisation" reduce compilation time
- You can reuse a compiled design multiple time during a single JVM execution, usefull to run other tests on the same design

wswslzp, 2020-09-08:
Thanks for reply 👍 
I am wondering whether this feature may be available in the future version?

Dolu1990, 2020-09-08:
So, technicaly speaking, it shouldn't be difficult, basicaly, just caching the shared object would avoid both verilator generation and compilation.

That isn't realy in my todo list for now, but is totaly faisable.

closed 2020-11-10 by wswslzp