SpinalHDL/SpinalHDL #354: the result of LatencyAnalysis is wrong when the path contians MemReadSync.

pdaxian, 2020-11-12:
I use the LatencyAnalysis to get the latency from input to output,
when the path contains a mem readAddr to readDataOut,  the latency I think is wrong.

```
import spinal.core._
import spinal.lib._

class LatencyTest extends Component {
  val io = new Bundle{
    val i1,i2 = in Bits(4 bits)
    val o1,o2 = out Bits(4 bits)
  }

  io.o1 := RegNext(io.i1)
  println("the latency of RegNext is : " + LatencyAnalysis(io.i1,io.o1))

  io.o2 := Mem(Bits(4 bits),16).readSync(io.i2.asUInt)
  println("the latency of mem readSync is : " + LatencyAnalysis(io.i2,io.o2))
}

object LatencyTestVerilog{
  def main(args: Array[String]): Unit = {
    SpinalConfig(targetDirectory = "./rtl/").generateVerilog(new LatencyTest)
  }
}
```

this print:

```
the latency of RegNext is : 1
the latency of mem readSync is : 0
```

pdaxian, 2022-04-07:
this is fixed at v1.6.4 or an earlier version, close issue.

```
[Runtime] SpinalHDL v1.6.4    git head : 598c18959149eb18e5eee5b0aa3eef01ecaa41a1
[Runtime] JVM max memory : 1990.0MiB
[Runtime] Current date : 2022.04.07 11:58:49
[Progress] at 0.000 : Elaborate components
the latency of RegNext is : 1
the latency of mem readSync is : 1
[Progress] at 0.402 : Checks and transforms
[Progress] at 0.555 : Generate Verilog
[Done] at 0.662
```

closed 2022-04-07 by pdaxian

Dolu1990, 2022-04-08:
Hooo, seems like that issue was lost in time XD
Thanks for the update ^^