SpinalHDL/SpinalHDL #356: BlackBox adds extra clock signal

kleinai, 2020-11-18:
I'm trying to instantiate a black box but an extra clock signal (in my case `clk`) is added after code generation.
My project is using `1.4.2`.

Black box component:
```
class MyBlackBox extends BlackBox {
  val io = new Bundle {
    val CLKI   = in Bool
    val RST    = in Bool
    val CLKOP  = out Bool
    val CLKOS  = out Bool
    val CLKOS2 = out Bool
    val CLKOS3 = out Bool
    val LOCK   = out Bool
  }
  noIoPrefix()
  mapCurrentClockDomain(io.CLKI, reset = io.RST, resetActiveLevel = HIGH)
}
```

Which generates:
```
MyBlackBox myBlackBox_1 (
  .CLKI      (clk_1                ), //i
  .RST       (reset                ), //i
  .CLKOP     (myBlackBox_1_CLKOP   ), //o
  .CLKOS     (myBlackBox_1_CLKOS   ), //o
  .CLKOS2    (myBlackBox_1_CLKOS2  ), //o
  .clk       (myBlackBox_1_clk     ), //o
  .LOCK      (myBlackBox_1_LOCK    )  //o
);
```

kleinai, 2020-11-20:
I figured this out myself. I had set the name of CLKOS3 to `clk`.
Is this a bug that `setName` can change the name of the black box?

Dolu1990, 2020-11-20:
Hi,

Hoo it make sense then.
It isn't a bug, basicaly, "what you do is what you get", as setName on that pin can be used as a feature in other cases.

Why did you .setName("clk") ?

Dolu1990, 2020-11-20:
Basicaly, the setName will override the name of the given reference, if that reference is the signal from that blackbox, that override will also impact the blackbox io name

kleinai, 2020-11-20:
I had used `.setName` because I curious if I could make the output a little more human readable. Maybe that's the wrong mentality to have with this kind of system.

closed 2021-02-19 by kleinai