SpinalHDL/SpinalHDL #358: VHDL blackbox

cinderellaking, 2020-12-19:
I ran into some issues with the blackbox for VHDL. Here is my code:

//================================================================
//  Start of Scala code
//================================================================

import spinal.core._
import spinal.lib._

case class vhdl_blackbox (Data_Width : Int) extends BlackBox {
    addGeneric("Data_Width", Data_Width)

    val io = new Bundle {
        val enable_in = in Bool
        val data_out	    = out Bits (Data_Width bits)
    }
}

case class vhdl_blackbox_test () extends Component {
    val io = new Bundle {
        val enable_in = in Bool
        val data_out_16 = out Bits (16 bits)
        val data_out_32 = out Bits (32 bits)
    }

    val blackbox16 = vhdl_blackbox(16)
    val blackbox32 = vhdl_blackbox(32)
    blackbox16.io.enable_in := io.enable_in
    blackbox32.io.enable_in := io.enable_in

    io.data_out_16 := blackbox16.io.data_out
    io.data_out_32 := blackbox32.io.data_out

}

object vhdl_blackbox_test{
  def main(args : Array[String]): Unit = {
      val report = SpinalVhdl(new vhdl_blackbox_test())
  }
}
//================================================================
// End of Scala code
//================================================================

And I ran it under spinalHDL 1.4.2, and the VHDL code generated is like the following:

//================================================================
// Start of VHDL code (generated)
//================================================================

......

entity vhdl_blackbox_test is
  port(
    io_enable_in : in std_logic;
    io_data_out_16 : out std_logic_vector(15 downto 0);
    io_data_out_32 : out std_logic_vector(31 downto 0)
  );
end vhdl_blackbox_test;

architecture arch of vhdl_blackbox_test is
    signal blackbox16_io_data_out : std_logic_vector(15 downto 0);
    signal blackbox32_io_data_out : std_logic_vector(31 downto 0);

  component vhdl_blackbox is
        generic( 
              Data_Width : integer  
        );
       port( 
          io_enable_in : in std_logic;
          **io_data_out : out std_logic_vector(15 downto 0)** 
       );
       end component;
  
begin
  blackbox16 : vhdl_blackbox
    generic map( 
      Data_Width => 16 
    )
    port map ( 
      io_enable_in => io_enable_in,
      io_data_out => blackbox16_io_data_out 
    );
  blackbox32 : vhdl_blackbox
    generic map( 
      Data_Width => 32 
    )
    port map ( 
      io_enable_in => io_enable_in,
      io_data_out => blackbox32_io_data_out 
    );
  io_data_out_16 <= blackbox16_io_data_out;
  io_data_out_32 <= blackbox32_io_data_out;
end arch;

//================================================================
// End of VHDL code (generated)
//================================================================

As you can see in the bold part of the VHDL code, the port length is hard coded to the length of the first instance in component declaration. But it actually should use the generic, like 

component vhdl_blackbox is
        generic( 
              Data_Width : integer  
        );
       port( 
          io_enable_in : in std_logic;
          **io_data_out : out std_logic_vector(Data_Width-1 downto 0)** 
       );
end component;

As the 2nd instance of the above code uses a different generic setting, the generate VHDL code will fail to compile.

Dolu1990, 2020-12-19:
Hooo shit right XD

So, 

```
component vhdl_blackbox is
generic(
Data_Width : integer
);
port(
io_enable_in : in std_logic;
io_data_out : out std_logic_vector(Data_Width-1 downto 0)
);
end component;
```
Would be a bit a paine in the ass to implement.

Do you think the following woul be fine ?

```
component vhdl_blackbox is
port(
io_enable_in : in std_logic;
io_data_out : out std_logic_vector() //Uninferred size
);
end component;
```

cinderellaking, 2020-12-19:
I tried the "Uninferred size" **as-is** under modelsim:    

vcom -2008 ../source/spinalHDL/vhdl_blackbox_test.vhd

And it came back as a syntax error:

Error: ../source/spinalHDL/vhdl_blackbox_test.vhd(406): near ")": syntax error

The synthesis tool (synplify pro) also thinks it is an error: "Expecting expression"

**However, if the parenthesis is removed**, like

component vhdl_blackbox is
port(
io_enable_in : in std_logic;
io_data_out : out std_logic_vector  //Uninferred size without parenthesis
);
end component;

then both modelsim and synplify pro can pass. 

So I guess that is the way to go.

closed 2020-12-22 by Dolu1990

cinderellaking, 2020-12-22:
Thank you so much for the fix. Happy holidays!

commit 3733ecf3 (2020-12-22): Fix #358 (VHDL blackbox generic/signal width)