SpinalHDL/SpinalHDL #359: $past is emited outside of a clocked block

Henkru, 2020-12-22:
## SpinalHDL version
```
1.4.2
```
## Issue
The following component
```scala
package mylib

import spinal.core._
import spinal.core.Formal._

class MyTopLevel extends Component {
  val counter = Reg(UInt(8 bits)) init(0)

  counter := counter + 1

  GenerationFlags.formal {
    when(initstate()) {
      assume(clockDomain.isResetActive)      
    }.otherwise {
      assert(past(counter) + 1 === counter)
    }
  }
}

object MyTopLevelVerilog {
  def main(args: Array[String]) {
    val config = SpinalConfig(defaultConfigForClockDomains = ClockDomainConfig(resetKind=SYNC, resetActiveLevel=HIGH))
    config.includeFormal.generateSystemVerilog(new MyTopLevel())
  }
}
```
generates:
```systemverilog
module MyTopLevel (
  input               reset,
  input               clk
);
  wire       [7:0]    _zz_1;
  reg        [7:0]    counter;

  assign _zz_1 = ($past(counter, 1) + 8'h01);
  always @ (posedge clk) begin
    if(reset) begin
      counter <= 8'h0;
    end else begin
      counter <= (counter + 8'h01);
      if($initstate())begin
        assume(reset);
      end else begin
        assert((_zz_1 == counter));
      end
    end
  end
endmodule
```
The `$past` function is only allowed in the clocked block as the output of SymbiYosys shows:
```
SBY 17:47:13 [fv_bmc] base: MyTopLevel.sv:14: ERROR: System function \$past is only allowed in clocked blocks.
```
If I move the following line inside the clocked block everything works.
```
assign _zz_1 = ($past(counter, 1) + 8'h01);
```
## Workaround
Changing
```scala
assert(past(counter) + 1 === counter)
```
to
```scala
assert(past(counter) === counter - 1)
```
emits the past function inside of the clocked block.

Dolu1990, 2021-01-01:
Hmm make sense

This is quite enoying because depending how you compose the expression tree, it realy need to place it in a combinatorial manner outside the clocked process.

What's about instead using past(xx) using RegNext(xx) ?

Henkru, 2021-01-03:
Yep, I understand that this is kind of rare corner case. Therefore RegNext is a good enough workaround for this.

piegamesde, 2021-06-21:
I've run into this too, and with seemingly normal code (also it seems there has been a regression, though I haven't bisected it yet).

Spinal:

```scala
when(past(clockDomain.isResetActive)) {
  assume(inStream.valid === False)
  assert(outStream.valid === False)
}
```

Verilog (I hope I've copied the right lines; this isn't minimized):

```verilog
assign when_SlowdownTest_l89 = $past((! resetn), 1);

always @(posedge clk) begin

if(when_SlowdownTest_l89) begin
  assume((_zz_when_SlowdownTest_l39 == 1'b0));
  assert((_zz_when_SlowdownTest_l98 == 1'b0));
end
```

Basically the condition of the `when` clause, `when_SlowdownTest_l89`, is combinatorial so it is put outside of the clocked block. But because of the `$past` command, this is kind of not correct.

Dolu1990, 2021-06-22:
What's about i implement the workaround (RegNext) in the SpinalHDL past function ? That way we do not rely on the $past macro  ?

piegamesde, 2021-06-22:
How exactly would that be implemented? Because I can't tell if/how this would resolve the issue. But sure, you can make a PR or branch for us to try out.

jiegec, 2021-06-22:
FWIW, this is how I solve the problem: https://github.com/jiegec/EspinalLib/blob/3a95e0c629b158b0cc62e683339afbeb6563f0ea/src/main/scala/espinallib/formal/package.scala#L72-L118. I got the inspiration from ZipCPU formal verification posts.

piegamesde, 2021-06-23:
@jiegec I'm sorry, but I don't really understand what that code does, and how and why.

@Dolu1990 Do you remember any specific change (or point in time) where the generated Verilog was changed such that the conditions of a `when` statement got factored out to a variable with an `assign` statement?

Regarding the `RegNext` workaround, I think this would only work for `past(…, delay = 1)` statements. Higher delays could of course be modeled by nesting registers, but it get ugly quickly. The other difference between `past` and `RegNext` are their behavior in the first cycle after reset. We'd have to make sure they have 100% equivalent semantics before doing such a drop-in replacement.

Dolu1990, 2021-06-24:
Hi,

Fixed in dev. So one thing about dev,  in Bool    out Bool    Bool need to be replaced with  in Bool()    out Bool()    Bool()

else all should be fine

Also, some feature were added to help with formal verification : 

```scala
class TopLevel extends Component {
  val io = new Bundle {
    val ready = in Bool()
    val valid = out Bool()
  }
  val valid = RegInit(False)

  when(io.ready) {
    valid := False
  }
  io.valid <> valid
  // some logic

  import spinal.core.GenerationFlags._
  import spinal.core.Formal._

  GenerationFlags.formal {
    assumeInitial(clockDomain.isResetActive) //Keep that statement simple to help the generation
    assert(!(valid.fall && !io.ready))

    ClockDomain.current.duringReset {
      assume(io.ready === False)
    }
  }
}

object MyToplevelSystemVerilogWithFormal {
  def main(args: Array[String]) {
    val config = SpinalConfig(defaultConfigForClockDomains = ClockDomainConfig(resetKind=SYNC, resetActiveLevel=HIGH))
    config.includeFormal.generateSystemVerilog(new TopLevel())
  }
}
```

will generate 

```verilog
module TopLevel (
  input               io_ready,
  output              io_valid,
  input               reset,
  input               clk
);
  reg                 valid;
  reg                 valid_regNext;

  initial begin
    assume(reset);
  end
  assign io_valid = valid;
  always @(posedge clk) begin
    if(reset) begin
      valid <= 1'b0;
    end else begin
      if(io_ready) begin
        valid <= 1'b0;
      end
      assert((! (((! valid) && valid_regNext) && (! io_ready))));
    end
  end

  always @(posedge clk) begin
    valid_regNext <= valid;
  end

  always @(posedge clk) begin
    if(reset) begin
      assume((io_ready == 1'b0));
    end
  end

endmodule
```

Those addtion would be able to superseed the $initialState i guess ? right ?

piegamesde, 2021-06-24:
Um, I am totally confused by that commit and comment? You sure you hit the correct issue? ^^

Dolu1990, 2021-06-24:
Ahhh  damned, sorry XD

So the $past realy need a replacement via the Delay(that, cycles). Can't enforce it to be emited in a clocked process.

My guess is that there is not much usages where cycles is higher than 1 ?
About the state durring reset, what is it for $past ? randomized ?

Dolu1990, 2021-06-25:
I pushed a workaround in dev, let's me know if you have issues ^^

closed 2021-06-25 by Dolu1990

piegamesde, 2021-06-25:
Btw this issue applies to `$past`, `$stable`, `$changed`, `$rose`, and `$fell` as well.

https://symbiyosys.readthedocs.io/en/latest/verilog.html

> … that there is no initial value given to any $past(<expr>). That means that on the first clock period of any design, $past(<expr>) will be undefined.

-----

There are a few use cases for `$past` with longer delays, but I'll admit one encounters them only rarely.

Dolu1990, 2021-06-25:
So, my guess is we should workaround them all, as it isn't a complicated thing.

Dolu1990, 2021-06-28:
done

commit 5a848ccd (2021-06-24): Fix #359
commit 53dcc404 (2021-06-25): Fix #359 ($past formal) by inferring register delays
commit 9f86e95c (2021-06-28): Fix #359