SpinalHDL/SpinalHDL #360: Formal statement emitted without trailing ';'

tomverbeure, 2021-01-04:
I added a formal assert to my code, and it created the following:

```verilog
    `ifndef SYNTHESIS
      `ifdef FORMAL
        assert((_zz_FirEngine_46_ <= _zz_FirEngine_47_))
      `else
        if(!(_zz_FirEngine_46_ <= _zz_FirEngine_47_)) begin
          $display("FAILURE ");
          $finish;
        end
      `endif
    `endif
```

There seems to be a ';' missing at the end of the assert?

I think this needs to be fixed here:
https://github.com/SpinalHDL/SpinalHDL/blob/e95483d53f93a650540d7fe2ba18baeb6f0524c2/core/src/main/scala/spinal/core/internals/ComponentEmitterVerilog.scala#L537

Dolu1990, 2021-01-04:
Fixed thanks ^^

Dolu1990, 2021-01-04:
(in dev)

commit c2bc59b1 (2021-01-04): fix #360 verilog assert semicolon
closed 2021-01-04 by Dolu1990