SpinalHDL/SpinalHDL #362: AXI4-Stream implementation

MarekPikula, 2021-01-08:
Hi, thanks for great work. Are there any plans to implement AXI4-Stream? I think it would be a really nice to have addition to family of buses and looking at Stream class it'd be probably very easy to implement some basic AXI Stream infrastructure. 

SpinalHDL serves me very well as means of making Vivado-block-design-style top-level blocks, but it lacks AXI Stream support, which I use in some blocks.

Dolu1990, 2021-01-11:
Hi

Currently, i'm personnaly not planning much about Axi stream, because i'm mostly used raw Stream only.

Else, i have to say, there is a similar thing to AxiStream named BSB : 
https://github.com/SpinalHDL/SpinalHDL/blob/dev/lib/src/main/scala/spinal/lib/bus/bsb/Bsb.scala

There is a few converter implemented, but no decoder / arbiter implemented.

likewise, 2022-11-16:
The existing Stream interface matches the AXI Stream interface quite well. Also with Stream(Fragment()) you can have framed/packetized AXI Streams.

Here is a project where I use Spinal lib Stream class to process AXI Streams with Framing (TLAST):

https://github.com/likewise/SpinalCorundum/tree/main/src/main/scala/corundum

Let me know what you think is missing?

MarekPikula, 2022-11-16:
When I created this issue, there was no AXI4-Stream implementation in Spinal. I guess it can be closed now, as I can see that the current implementation is complete (from my perspective).

closed 2022-11-17 by Readon