SpinalHDL/SpinalHDL #370: Signal Visibility in Simulation

distributed, 2021-01-22:
As already discussed on Gitter, here is an example component that does a couple of translations to a Vector of Streams. 

Ideally I would like all signals stages1-stages5 to be visible in simulation, however I can only see stages 1, 3 and 5. 

I am unsure whether `stages2` even can be made visible in any meaninful way. I hope you can tell me about what to expect here :)

In my intuition, `stages4` and `stage4Occupancies` should both be visible, however they are not.

I hope the example is not too maximal, but it's easy to strip away stuff. I wanted to include some working stuff as well to show what is behaving as expected and what note.

I am using SpinalHDL v1.4.2,

```
package repro

import spinal.core.sim._
import spinal.core._
import spinal.sim._
import spinal.lib._

class SignalNaming extends Component {
    val numStreams = 3

    val io = new Bundle {
        val input = Vec(slave Stream(Bits(8 bits)), numStreams)
        val output = Vec(master Stream(Bits(10 bits)), numStreams)

        val ctrl1 = in Bool
    }

    val stages1 = io.input.map(_.takeWhen(io.ctrl1))

    // Nop, renaming the signals. might be because i renamed a signal.
    // It might be that there's no reasonable way to support having
    // signals visible by both their stages1 and stages 2 names.
    val stages2 = stages1

    val stages3 = stages1.map(stream => {
        val counter = Reg(UInt(3 bits))
        when (stream.fire) {
            counter := counter + 1
        }

        val inced = stream ~~ (_.asUInt.resize(9 bits) + counter)

        inced ~~ (pl => pl(0, 3 bits) @@ pl(3, 6 bits))
    })

    val (stages4, stage4Occupancies) = stages3.map(stream => {
        val fifo = new StreamFifo(UInt(9 bits), 16)
        stream >> fifo.io.push
        (fifo.io.pop, fifo.io.occupancy)
    }).unzip

    val clamp = (stage4Occupancies.reduce(_+_) === 0)
    val stages5 = stages4.map(_ ~~ (d => Mux(clamp, U(0), d)))

    val preOutput = stages5.map(_ ~~ (_.resize(10 bits).asBits))

    (preOutput, io.output).zipped.foreach(_ >> _)
}

object SignalNaming {
    def main(args: Array[String]): Unit = {
        val compiledSim = SpinalSimConfig().withWave.compile(new SignalNaming)

        compiledSim.doSimUntilVoid("signalnaming") { dut =>
            dut.clockDomain.forkStimulus(10)
            dut.clockDomain.waitSampling(1)
            simSuccess()
        }
    }
}
```

Dolu1990, 2021-01-24:
```scala
   val stages2 = stages1.combStage()

  (fifo.io.pop.combStage(), CombInit(fifo.io.occupancy))
```

Should do it ^^

distributed, 2021-01-25:
Thanks, that does the trick! :)

Is that issue closed or is there something that remains to be done in Spinal from your point of view?

Dolu1990, 2021-01-25:
hmm, ultimately, that could be something for SpinalHDL to handle automaticaly, not in the same way, but by creating dummy copy of the signals which are already named, only for simulation usages.

We can keep the issue.

Dolu1990, 2021-01-25:
(VHDL alias feature)

Readon, 2023-01-04:
Would the simPublic method helps? If yes, I think this issue could be closed.

numero-744, 2023-01-05:
Might sound crazy: has anyone considered contacting Synopsys to know if they would be interested in testing that interface, or participate in its implementation by any way?

Said another way; to ask them if they are "interested into bringing to their tools better support for a language that let their clients improve their productivity, so that both Synopsys and their clients can be more competitive".

numero-744, 2023-01-05:
So in this case I think `CombInit` is a good solution. It will be documented as we have already discussed earlier.

@distributed if it is good for you, can we close this issue?

distributed, 2023-01-06:
I'm currently using `CombInit` and friends whenever the problem pops up, which is only a minor annoyance and works well.

For me we can close the issue, but I think @Dolu1990 wanted to keep it open (see Jan 25, 2021) as a reminder.

numero-744, 2023-01-06:
Ah to add a build hook to intercept new `val` signals; when the RHS is already named, create a new signal instead of just aliasing? Sounds to be a breaking change because aliasing will likely not work anymore, unless the new signal is just a copy with open output (no drivees/fanout)? But in this case it might trigger "unused signal" warnings in simulators?

Readon, 2023-01-06:
> I'm currently using `CombInit` and friends whenever the problem pops up, which is only a minor annoyance and works well.
> 
> For me we can close the issue, but I think @Dolu1990 wanted to keep it open (see Jan 25, 2021) as a reminder.

Then it's done through reserve the name policies, so the solution could be documented into [Nameing](https://spinalhdl.github.io/SpinalDoc-RTD/dev/SpinalHDL/Structuring/naming.html) site.

Dolu1990, 2023-01-06:
> Ah to add a build hook to intercept new val signals; when the RHS is already named, create a new signal instead of just aliasing?

As long as the lhs is still the reference to the rhs, and not a newly created signal.
Also, maybe better to have things stored as alias in the ast, instead of creating new signal.