SpinalHDL/SpinalHDL #388: JTAG TAP compatibility wrt IDCODE

tomverbeure, 2021-03-09:
While playing with the JTAG logic, I ran into the following compliance incompatibilities:

- **when an IDCODE IR chain is defined, the instruction register should select this IR chain when coming out of reset, or after entering the Test-Logic-Reset TAP state** (IEEE 1149.1-2001, section 8.13.2). When there is no IDCODE register, then the default IR chain should be BYPASS.
Right now, the TAP comes out of reset with IR chain 0. Doing this right is really important for auto-discover: if all devices scan out an IDCODE by default, then a tool like OpenOCD can figure out all the JTAG TAPs in the system with a single DR-SCAN that doesn't require an initial IR-SCAN. I know that the instruction register is defined [here](https://github.com/SpinalHDL/SpinalHDL/blob/3097aa38e8419acac302cfacff70f1cd700b1865/lib/src/main/scala/spinal/lib/com/jtag/JtagTap.scala#L75), but I don't know now to give it a reset value that depends on whether or not an IDCODE has been defined?

- **the IDCODE must be loaded into the scan chain when the TAP enters the Capture-DR state.** (section 8.13.1) 
Right now, the IDCODE is only loaded during the RESET state. As a result, when scanning out the IDCODE a second time without going through RESET, you get the previously scanned in value instead. I think the IDCODE is loaded [here](https://github.com/SpinalHDL/SpinalHDL/blob/3097aa38e8419acac302cfacff70f1cd700b1865/lib/src/main/scala/spinal/lib/com/jtag/JtagTap.scala#L115), and that the fix is as simple as replacing `JtagState.RESET` by `JtagState.DR_CAPTURE`.

Dolu1990, 2021-03-11:
Hi ^^

> IDCODE IR

Right currently it was always assumed that there would be the IDCODE in the tap. 
Ahh then about to having no idcode, maybe we should add a function on the tape itself to specify the "idcode less" behavour, then user need to call it and specify the reset value of the IR.
So similarly to https://github.com/SpinalHDL/SpinalHDL/blob/b6ae0a74d2d5838d16fadb6978da77d5e9709ec1/lib/src/main/scala/spinal/lib/com/jtag/JtagTap.scala#L112, having a "def noidcode"

> the IDCODE must be loaded into the scan chain when the TAP enters the Capture-DR state.

I'm not sure to understand, i mean, IDCODE is an instruction like all others, at the exception that the instruction register is set to IDCODE when a reset occure, isn't it ?
The mistake i can see in the current implementaion is there i would say : 
https://github.com/SpinalHDL/SpinalHDL/blob/b6ae0a74d2d5838d16fadb6978da77d5e9709ec1/lib/src/main/scala/spinal/lib/com/jtag/JtagTapInstructions.scala#L115
I only load the shift register in the reset condition, but it should have been done in the capture-dr state i think.

Make sense ?

> during CAPTURE-IR, the 2 LSBs of the IR shift register should be set to "01".

But then, how can the jtag master read in which instruction the tap is ? it can't ?

tomverbeure, 2021-03-11:
> Ahh then about to having no idcode, maybe we should add a function on the tape itself to specify the "idcode less" behavour, then user need to call it and specify the reset value of the IR.

When no IDCODE is specified, then BYPASS should be selected when coming out of reset. And BYPASS should always have an IR code with all bits set to 1.

> I only load the shift register in the reset condition, but it should have been done in the capture-dr state i think.

Correct. That's what I've done in my fix.

> But then, how can the jtag master read in which instruction the tap is ? it can't ?

That's right: it can't!

Dolu1990, 2021-03-16:
Then it sound all good to me ^^

You were able to test that on hardware with VexRiscv ? Via a JTAG tap made in the FPGA fabric ?

> When no IDCODE is specified, then BYPASS should be selected when coming out of reset. And BYPASS should always have an IR code with all bits set to 1.

That one i forgot XD

tomverbeure, 2021-03-16:
> You were able to test that on hardware with VexRiscv ? Via a JTAG tap made in the FPGA fabric ?

Yes.

Tom

Dolu1990, 2021-03-18:
Merged, thanks :D

Dolu1990, 2021-03-18:
@tomverbeure This may interrest you : 
https://github.com/SpinalHDL/VexRiscv/tree/dev#fpu

tomverbeure, 2021-03-18:
> @tomverbeure This may interrest you :
> https://github.com/SpinalHDL/VexRiscv/tree/dev#fpu

I’ve been watching it with interest!