SpinalHDL/SpinalHDL #390: User defined stream io error

dahuwei, 2021-03-12:
I'm trying to define new stream based on Stream(Fragment(T)),  it works well when I use master Stream Fragment(MyStreamData(32)) for io.output, but it doesn't work when I use MyStream object to replace it. 
It works well when I remove master and call io.output.asMaster later. 
How can I make the MyStream works in the io bundle ?

[image]

volatile-static, 2021-03-13:
Why don't ask in our WeChat group? 😂

Dolu1990, 2021-03-13:
@dahuwei 
It is a limitation of the syntax implemented in SpinalHDL.
Basicaly, master Stream(xxx) is a shortcut of master(Stream(xxx)), but don't work if you have anything else than Stream or Flow.

So in your case, you have to use the long way  "master(MyStream(32))" to have it working.

dahuwei, 2021-03-13:
@Dolu1990 
Thanks, "master(MysStream(32))" works.
I just started tring to use spinal to generate verilog code, and I think spinal is really a great job.

dahuwei, 2021-03-13:
> Why don't ask in our WeChat group? 😂

@volatile-static 
I'm not in that group, can you share the QR coe ?

volatile-static, 2021-03-13:
https://mp.weixin.qq.com/s/fLyV2V8Rf-o8INF7AmlObg
@dahuwei  Welcome~

closed 2022-10-23 by numero-744