SpinalHDL/SpinalHDL #394: Axi4Decoder: WriteOnly: missing a conditional check to avoid deref NULL pointer

lastweek, 2021-03-25:
Hi @Dolu1990,

I'm using the `Axi4Crossbar` and found when the slave address mapping covers the whole space, there will be a NULL pointer error in `Axi4WriteOnlyDecoder`.

Using whole address space is enforced due to this line: https://github.com/SpinalHDL/SpinalHDL/blob/dev/lib/src/main/scala/spinal/lib/bus/amba4/axi/Axi4Decoder.scala#L29

The error line is:
https://github.com/SpinalHDL/SpinalHDL/blob/dev/lib/src/main/scala/spinal/lib/bus/amba4/axi/Axi4Decoder.scala#L106
```scala
io.input.writeData.ready := ((pendingSels & io.outputs.map(_.writeData.ready).asBits).orR || (pendingError && errorSlave.io.axi.writeData.ready)) && allowData
```

It should have a `if (decodingErrorPossible)` like this line does: https://github.com/SpinalHDL/SpinalHDL/blob/dev/lib/src/main/scala/spinal/lib/bus/amba4/axi/Axi4Decoder.scala#L33

Or am I missing something? Looking forward to your reply

---

My code
```scala
  val axiCrossbar = Axi4CrossbarFactory()
  axiCrossbar.addSlaves (
    parsers(0).io.axi   -> (0, 0x4000),
    parsers(1).io.axi   -> (0x4000, 0xC000)
  )
  axiCrossbar.addConnections (
    io.axi   -> List(parsers(0).io.axi, parsers(1).io.axi)
  )
  axiCrossbar.build()
```

The error msg:
```
ERROR !                                             
A null pointer access has been detected in the JVM.                                                      
This could happen when in your SpinalHDL description, you access an signal which is only defined further.
For instance :                                      
  val result = Bool                                 
  result := a ^ b  //a and b can't be accessed there because they are only defined one line below (Software rule of execution order)
  val a,b = Bool                                    
                                                                                                
[info]   java.lang.NullPointerException:            
[info]   at spinal.lib.bus.amba4.axi.Axi4WriteOnlyDecoder.<init>(Axi4Decoder.scala:97)                   
[info]   at spinal.lib.bus.amba4.axi.Axi4CrossbarFactory$$anonfun$2$$anon$2.<init>(Axi4Crossbar.scala:142)                               
[info]   at spinal.lib.bus.amba4.axi.Axi4CrossbarFactory$$anonfun$2.apply(Axi4Crossbar.scala:138)        
[info]   at spinal.lib.bus.amba4.axi.Axi4CrossbarFactory$$anonfun$2.apply(Axi4Crossbar.scala:124)        
[info]   at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:234)             
[info]   at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:234)             
[info]   at scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)               
[info]   at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)                           
[info]   at scala.collection.TraversableLike$class.map(TraversableLike.scala:234)                        
[info]   at scala.collection.AbstractTraversable.map(Traversable.scala:104)
```

Dolu1990, 2021-03-27:
Hi,

Yes you are right, those errorSlave usages should not be executed if there no error slave ^^

Do you feel ok to make a PR ?

lastweek, 2021-03-28:
> Hi,
> 
> Yes you are right, those errorSlave usages should not be executed if there no error slave ^^
> 
> Do you feel ok to make a PR ?

yeah, i will do a PR in a day or two.

lastweek, 2021-03-29:
Related PR: https://github.com/SpinalHDL/SpinalHDL/pull/396

closed 2021-04-01 by lastweek