SpinalHDL/SpinalHDL #398: Is there any why to print the value of a signal rather than it's name?

ArcheyChen, 2021-04-01:
@Dolu1990 
Hello, I try to add reports to my code,like this:

```scala
  when(nestUpdate.fire){
    assert(nestUpdate.set === status.set && nestUpdate.tag === status.tag && nestUpdate.clientID === io.MSHRID)
    dirData.state := nestUpdate.newState
    report(s"dir state in MSHR#${io.MSHRID} change to ${io.nestUpdate.newState}. By NestUpdate")
  }
```
I want to print the status change every time the nestUpdate is fire

but the generated verilog code is like this:
``` verilog
`ifndef SYNTHESIS
          `ifdef FORMAL
            assert(1'b0);
          `else
            if(!1'b0) begin
              $display("NOTE dir state in MSHR#(toplevel/io_MSHRID : in UInt[1 bits]) change to (toplevel/io_nestUpdate_payload_newState : in SpinalEnumCraft). By NestUpdate");
            end
          `endif
        `endif
```

it looks like spinal is trying to print the NAME of signal , instead of it's value.

but in chisel ,they use the funtion" printf " to do such thing:
```scala
val myUInt = 33.U
printf(p"myUInt = $myUInt") // myUInt = 33
printf(p"myUInt = 0x${Hexadecimal(myUInt)}") // myUInt = 0x21
printf(p"myUInt = ${Binary(myUInt)}") // myUInt = 100001
printf(p"myUInt = ${Character(myUInt)}") // myUInt = !
```
I know that I can use something like xx.toInt in spinal simulation.But want to insert this kind of note in my module. --- I can see it when i using verilog to do simulation.
is there any way to do so?

ArcheyChen, 2021-04-02:
@Dolu1990 
Hello, could you add a primitive similar to display in spinal? such as
display("data is ：%d",mydata)
Just translate mydata into the variable name in the final Verilog to solve the problem.
True, now in testbench, you can pull the signal in the module and output it with the function println of scala. But in a complex design, it often increases the complexity of writing code. For example, if the signal needs to be output under a complex condition, I need to rewrite the judgment condition in the top-level testbench
It's like writing the following code in Verilog:
```verilog
   always @ (posedge clk) begin
      if(dut.submodule.value1 == 1 && dut.submodule.value2 == 3 &&dut.submodule.value3 == 3 ……)
            $display("the data is :%d",dut.submodule.mydata);
   end
```
I believe that adding such a function wrong be too hard, but will be useful in lots of cases
-------
My graduation project is written by spinal HDL. It's really fast and comfortable to develop spinal, but now I need to verify my code, and the lack of this function really bothers me. I need this function very much, otherwise I may face the difficulty of not completing the graduation project on time.
I hope you can reply me when you are free. Thank you.

Dolu1990, 2021-04-03:
Hi,

So there is something similar, not as polished than than using the p"", but functional : 

```scala
object Play1 extends App{
  import spinal.core.sim._
  SimConfig.compile(new Component {
    val a = in UInt(8 bits)
    val a2 = a+a
    report(Seq("miaou ", a, " rawrrr ", a2, " miaouuu"))
  }).doSim{ dut =>
    dut.clockDomain.forkStimulus(10)
    for(i <- 0 until 10){
        dut.a #= i
      dut.clockDomain.waitSampling()
    }
  }
}
``` 

``` 
[Progress] Start unnamed test simulation with seed 1076072331
[Done] Simulation done in 16.664 ms
NOTE miaou 00 rawrrr 00 miaouuu
NOTE miaou 01 rawrrr 02 miaouuu
NOTE miaou 02 rawrrr 04 miaouuu
NOTE miaou 03 rawrrr 06 miaouuu
NOTE miaou 04 rawrrr 08 miaouuu
NOTE miaou 05 rawrrr 0a miaouuu
NOTE miaou 06 rawrrr 0c miaouuu
NOTE miaou 07 rawrrr 0e miaouuu
NOTE miaou 08 rawrrr 10 miaouuu
```

You can put that report statement in when statements to conditionaly report it ^^

let's me know if you ahve issues with that feature, i implemented it but personnaly never used it. Seems to work well so far.

Dolu1990, 2021-04-03:
This may interrest you, i added  a smooth syntax.
https://github.com/SpinalHDL/SpinalHDL/commit/1f2aa47eb7dc761aaa05242946a19926a3f7218d

ArcheyChen, 2021-04-05:
Thanks this really helps!

ArcheyChen, 2021-04-06:
@Dolu1990 
Hello, I found that if I try to use `report(Seq("ReqType is","io.newReq"))` to print a enum value. It will print as a normal hex value (like 0,1)

but in verilog , there is a reg call "io_newReq_string" which content the name of enum value.

is there a way to print the name string instead of hex value?

Dolu1990, 2021-04-08:
True ^^
Fixed now.

ArcheyChen, 2021-04-09:
> True ^^
> Fixed now.

Thanks, but what about printing value in dec or binary? and to print the simulation time?(help to locale the time in wave form)

ArcheyChen, 2021-04-09:
I try to add  `case m: Bits => "%b"` to `ComponentEmitterVerilog.scala `
then I can print out one hot code easily

as for  printing value in Decimal , I can't modify the code because I dont know lots about the code of Spinal.

then I found that I can call `report(a.toBools)` to print a on hot code, but when I use `report(L"${a.toBools}")` it will fail, it seems like `L"something ${value}"` can't handle a Seq type of data

so ,in conclusion

1.there is a **BUG** in `report(L"something")` syntax, it can't handle a Seq type of data
2.need some syntax like: `report(L"dec value is ${a.printDec}, oct value is ${a.printDec}, binary value is ${a.printBin}")`
3.also in simulation ,need some syntax like `report(L" ${ClockDomain.getTime}")"`

Dolu1990, 2021-04-16:
3) added in branch i'm working on
2) maybe the best would be to follow the scala f"${42}%d" syntax ?
1)  Basicaly it can only handle BaseType for now. We could try to emulate things at elaboration time, but it seems too specific to me.

commit fd4e89b0 (2021-04-08): Fix #398
closed 2022-10-23 by numero-744