SpinalHDL/SpinalHDL #399: Mixed-width Ram problems

yindeliu, 2021-04-06:
when i try to use **mixed-width function** in **RAM/ROM**,error occurs and i have no idea about it.if it possible to give a demo code to demonstrate how to use **mem.writeMixedWidth** etc.(in the figure below) in spinalhdl document,thanks!;

below is my test code and the error message 

- my test code(implement a ram with 2 bits width write port and 4 bits width read port):

```
class MixedWidthRamTest extends Component{
  val io=new Bundle{
    val din=in Bits(2 bits)
    val addrWr=in UInt (4 bits)
    val addrRd=in UInt (3 bits)
    val dout=out Bits(4 bits)
  }

  val ram=Mem(Bits(4 bits),8)
  
  ram.writeMixedWidth(
    address = io.addrWr,
    data=io.din
  )
  io.dout:=ram.readSync(
    address = io.addrRd
  )

}
```

- Error Message:

```
Write data width (2 bits) is not the same than the memory one (ram) at
    mylib.MixedWidthRamTest.<init>(AddTest.scala:139)
    mylib.MyTopLevelVerilogWithCustomConfig$$anonfun$main$1.apply(AddTest.scala:171)
    mylib.MyTopLevelVerilogWithCustomConfig$$anonfun$main$1.apply(AddTest.scala:171)
    mylib.MyTopLevelVerilogWithCustomConfig$.main(AddTest.scala:171)
    mylib.MyTopLevelVerilogWithCustomConfig.main(AddTest.scala)
```
[image]

yindeliu, 2021-04-08:
@Dolu1990 could you give any support about that? thanks:)

Dolu1990, 2021-04-08:
Hi,

i remember that i had a similar issue recently, and it was solved.
It is in dev : 
https://github.com/SpinalHDL/SpinalHDL/commit/2000e36e61c966d88762b19ecddeaec15d6c6e59#diff-1c1a0c012fde8f34b5db6150c4418753c46fa0f07e4fa5ffd37835cfebaf9314

But not released in 1.4.3.

Can you give a try with SpinalHDL dev ? 
See https://spinalhdl.github.io/SpinalDoc-RTD/SpinalHDL/About%20SpinalHDL/faq.html#how-to-use-an-unreleased-version-of-spinalhdl-but-committed-on-git for how to use dev as if it was already released.