SpinalHDL/SpinalHDL #400: Feature request: read/write 'x' or 'z' signals in simulation

jiegec, 2021-04-07:
Icarus verilog supports x and z signals in simulation and it can be retrieved via VPI interface. Currently, the vpi plugin replaces them with 0 or 1 in random:

```cpp
void sanitize_byte_str(char* byte_str){
    for(size_t i = 0; i< 8; i++) {
        if ((byte_str[i] != '0') && (byte_str[i] != '1')) { 
            byte_str[i] = bin_dis(global_mt_rand) == 0 ? '0' : '1';
        }
    } 
}
```

However, it can be useful to read and write 'x' and 'z' to signals. It's good to have something like:

```
Bits/UInt/SInt.toBinString: Read a hardware BitVector as a binary Scala String value
Bits/UInt/SInt.toHexString: Read a hardware BitVector as a hex Scala String value
Bits/UInt/SInt #= B"010101xxxzzz"
Bits/UInt/SInt #= H"a1b2c3XZ"
```

jiegec, 2021-04-08:
Reading the code, there are two ways:

1. Add read/write_bin_str calls along read/write ones. This will not affect existing functions, but it needs lots of duplicated code.
2. Pass binary string between SharedMemIface and VpiPlugin, and persist original behavior in read/write functions. This might decrease performance.

Dolu1990, 2021-04-08:
Hi, 

Personnaly, i try to run away from any Z, as it isn't realy something from the digital world, but X make sense i think.
What's a third solution, where we have #= B"010101xxxxxx" in the SpinalHDL sim API which randomize all x bits  because giving a concrete binary value to the simulation backend ?

(This would keep everything in line with Verilator)

closed 2022-03-06 by jiegec