SpinalHDL/SpinalHDL #403: generated verilog all ports named '_zz_x'

dxsm, 2021-04-12:
Hi, I pull the latest SpinalHDL and tried a simple example, but the generated rtl looks strange like below:
[image]

And I tried the  V1.3.8, that is OK, but the later version all generate '_zz_x' port name instead of val defined name .It's so strange, can you help me with this problem? I used IDEA,sbt 1.3.3 and scala 2.11.12.

Dolu1990, 2021-04-12:
Hi,

that's a weird one XD

To update to the latest version, the build.sbt should also be updated a bit. especialy : 
https://github.com/SpinalHDL/SpinalTemplateSbt/blob/master/build.sbt#L9

Did you updated it ?

dxsm, 2021-04-12:
Actually I am trying the source code of dev branch，and the build.sbt also used the dev branch file: https://github.com/SpinalHDL/SpinalHDL/blob/dev/build.sbt

Dolu1990, 2021-04-13:
hoo right i can see that now on the screen shot.

Hmmm, can you send me a zip of your exact project setup ? That way i can try in similar conditions.
(That's reaily a weird issue, maybe it is environnement related)

dxsm, 2021-04-13:
This is my settings as the below attachment.
[settings.zip](https://github.com/SpinalHDL/SpinalHDL/files/6304288/settings.zip)

Dolu1990, 2021-04-13:
Sorry, i meant,  can you send me your whole project ?

dxsm, 2021-04-13:
Oh,I see, the whole project is too big, I create a temp repository. You can get from https://github.com/dxsm/Spinal.git
And my test scala file is https://github.com/dxsm/Spinal/blob/main/SpinalHDL/tester/src/test/scala/spinal/tester/scalatest/test.scala
The generated rtl is https://github.com/dxsm/Spinal/blob/main/SpinalHDL/Top.v

Dolu1990, 2021-04-13:
Got it solved by doing a sbt clean. Can you give a try ?

dxsm, 2021-04-14:
After doing a sbt clean, I encountered the error of "Object  Info is not the member of package Spinal.core", is there any file missing?
[image]

dxsm, 2021-04-14:
I tried to add the Info Object myself, there is no error now. But the generate rtl is the same as previous ”_zz_x“ ports.
[image]

Dolu1990, 2021-04-14:
ahhh, that's intellij bug this time XD
Basicaly, you need to do a sbt compile, it will  generate a few file that intellij forget to generate

dxsm, 2021-04-14:
Great！Got it solved by doing a sbt compile. Thanks a lot. 😀

dxsm, 2021-04-14:
solved

closed 2021-04-15 by Dolu1990