SpinalHDL/SpinalHDL #405: some views about SpinalHDL

dxsm, 2021-04-16:
After using spinalHDL for a while, No wonder this is a good alternative to HDL language. But I think there are still some things for improvement
1、The top thing annoying me is that the generated rtl have many "_zz_" signals, which prevents me making extensive use. I think an elegant rtl is very import，because for now, we cannot completely get rid of verilog/vhdl.
2、In my opinion, generate rtl should not have many scattered repetitive code. For example, sat and round will be large scale used in algorithm module, it will result in many scattered sat and round rtl in the module,. why not pack sat or round to a module and instantiate it? This will be more elegant.
3、Verilog fetures is not fully support：
a) generate for, generate if are not support which I think is a good feature to simplify code. Even though you can use scala for to generate code. But if I want to instantiate a module 100 times，it which copy 100 times. This is not elegant.
b) verilog data[start +: width] or data[start -: width], this type range select is no support. Often, I don't want to calculate the end point. This type is very helpful.

dxsm, 2021-04-17:
c)verilog parameter is not support. Parameter may help to reduce rtl files. You can imagine that saturate IP will be generated many files with out parameter,only because of the change of bit width. 
d) localparam is not support,which is good for fsm debug in verdi. This point has been mentioned in other issues.

Dolu1990, 2021-04-17:
Hi,

Thanks for the feedback :)

1) Yes it is a bit sad, that's mostly a issue with Verilog in which we can't define proper operator function to adapt the operator sementic between SpinalHDL and Verilog. In VHDL things are much more smooth. Often in the verilator backend i take the safe path of pushing things into explicit signal of a explicit size instead of relying on operations chaining. This can be improved in some cases, but i don't know verilog that much.
2) For me, the most important is that what you described in SpinalHDL is emited without area/fmax overhead. The paradigme which is used by SpinalHDL (using scala as a elaboration language to fill a raw netlist) work as if all the elaboration of the RTL was done by executing the scala program you write. No elaboration feature as if/for/paramters goes through the scala/SpinalHDL elaboration by design, also Verilog/VHDL elaboration feature are so limited it does not even worth a try i would say, it would mostly endup complicating things in a project with limited man power.
3)a) Partialy answered in point 2, also i would say at one moment we need to be pragmatic and not too "pedentic", i see what is generated by SpinalHDL as a netlist. The synthesis flow will anyway flatting things as SpinalHDL does. Then sure, human inspection of the netlist can be usefull, it never was an issue personnaly. What's about considering SpinalHDL as your human written hardware ? and Verilog/VHDL as stupid netlist ?
3)b) It is supported

```scala
object Play extends App{
  SpinalVerilog(new Component{
    val sel = in UInt(4 bits)
    val a = in Bits(32 bits)
    val result = out(a(sel, 8 bits))
  })
}
```

There is also some "better" alternative syntax as : a.subdivideIn(2 bits)(sel)   which will split a in parts of 2 bits and then mux it

3)c) thisis not supported it goes completly against the paradigme (scala feature used as elaboration feature), we could offcourse imagine some way / syntax to support it, but it would be a enginnering trap full of technological depth i gess.

3)d) right.

dxsm, 2021-04-21:
Mmm, generated by SpinalHDL as a netlist. It is a good idea if the EDA tools support scala for better debugging.

Dolu1990, 2021-04-21:
> Mmm, generated by SpinalHDL as a netlist. It is a good idea if the EDA tools support scala for better debugging. 

This will never happen.

When i mean netlist, i mean Verilog ^^

jijingg, 2021-09-11:
> 1、The top thing annoying me is that the generated rtl have many "zz" signals, which prevents me making extensive use. I think an elegant rtl is very import，because for now, we cannot completely get rid of verilog/vhdl.

This problem has made great progress in version 1.6

> 2、In my opinion, generate rtl should not have many scattered repetitive code. For example, sat and round will be large scale used in algorithm module, it will result in many scattered sat and round [...]

I suggest you use fixto wrapper,  this  can avoid generating fragmented code, The same fxipoint operation uses the same module
```scala
  class Top extends Component {
    val a = in SInt(8 bit)
    val b = out SInt(8 bit)
    //  b := a.round(2 bit).sat(1 bit)
    b := a.fixTo(6 downto 2, RoundType.ROUNDTOINF)
  }
```

> c)verilog parameter is not support. Parameter may help to reduce rtl files. You can imagine that saturate IP will be generated many files with out parameter,only because of the change of bit width.

This make sense. For modules with similar functions (MEM, FIFO, CORDIC cell), just some simple parameters, such as the change of bit width, will generate a new module, which is not elegant. However, there is a challenge that the parameter may not be able to check such as width mismatch. 
However, a class parameter (Val data: int) can be added to the spinal.core to complete the legitimacy check and replace  with parameter to avoid generating multiple similar .v files.  This function will be supported in the future, but the parameter usage should be very careful and Specific occasions.

dxsm, 2021-09-12:
@jijingg 
> This problem has made great progress in version 1.6

yeah,I see, it's great progress.

> I suggest you use fixto wrapper, this can avoid generating fragmented code, The same fxipoint operation uses the same module

Hoo, seems perfect, I'll try it.

> This make sense. For modules with similar functions (MEM, FIFO, CORDIC cell), just some simple parameters, such as the change of bit width, will generate a new module, which is not elegant. However, there is a challenge that the parameter may not be able to check such as width mismatch.
However, a class parameter (Val data: int) can be added to the spinal.core to complete the legitimacy check and replace with parameter to avoid generating multiple similar .v files. This function will be supported in the future, but the parameter usage should be very careful and Specific occasions.

In each module, parameter which specify the default value. When instance a module the parameter value will overlap the default value. I mean, the parameter always have a certain value, why not use the certain value to check the width mismatch. And generate the RTL use the literal as Enum. So I think it is possible to realize parameter width check.
I hope to define a `PInt` object to define a parameter Int, then you can use as this
```scala
class test(var width: PInt = PInt(8)) extends Component{  //generate a module with parameter
  val a = in SInt(width.value bit)
  val b = out SInt(width.value bit)
}

class top() extends Component{
  val inst1 = new test(PInt(16))
  val inst2 = new test()   // use defalut parameter
}
```

closed 2022-11-22 by numero-744