SpinalHDL/SpinalHDL #407: Why do the util.CountOne using lut grouped by 3?

JimmyJuan, 2021-04-19:
In `spinal.lib.Utils.scala`:

```scala
object CountOne{
  def args(thats : Bool*) : UInt = apply(thats)
  def apply(thats : BitVector) : UInt = apply(thats.asBools)
  def apply(thats : Seq[Bool]) : UInt = {
    if(thats.isEmpty) return U(0, 0 bits)
    val lut = Vec((0 until 1 << Math.min(thats.size, 3)).map(v => U(BigInt(v).bitCount, log2Up(thats.size + 1) bits)))
    val groups = thats.grouped(3)
    val seeds = groups.map(l => lut.read(U(l.asBits()).resized)).toSeq
    seeds.reduceBalancedTree(_+_)
  }
}
```

I don't understand why the inputs are grouped by 3, and then went through a lut.
As far as I know, different part of FPGA has different architecture, the size of LUT in CLB(slice) is different. I don't think it's a good idea to write it that way.

Dolu1990, 2021-04-19:
Hi,

Basicaly, i set it to 3 there to already have the functionality working.

I agree this is not greate, this parameter could be either a arguement of the function, either it could be loaded from the SpinalConfig used to generate the hardware.

Dolu1990, 2021-09-17:
Finaly implemented XD

Thanks for the issue :)

Dolu1990, 2021-09-17:
So before your SpinalVerilog / SpinalVhdl, you can do   LutInputs.set(6)   for instance, or LutInputs(6).on{ yourCode here }

closed 2021-09-17 by Dolu1990