SpinalHDL/SpinalHDL #408: assignFromBits behave strange

ArcheyChen, 2021-04-21:
```scala
class MyBundle extends Bundle{
  val a,b = UInt(8 bits)
  // or : val a = UInt(8 bits)
  //       val b = UInt(8 bits)
}
object Play2 extends App{
  import spinal.core.sim._
  SimConfig.compile(new Component {
    val a = UInt(8 bits)
    val b = UInt(8 bits)
    a := 1
    b := 2
    val myBundle = new MyBundle
    myBundle.assignFromBits(Cat(a,b))
    report(L"a=${myBundle.a} b=${myBundle.b}")
  }).doSim{ dut =>
    dut.clockDomain.forkStimulus(10)
    dut.clockDomain.waitSampling(5)
  }
}
```
and it outputs:
```
NOTE a=02 b=01
NOTE a=02 b=01
NOTE a=02 b=01
NOTE a=02 b=01
```
is this kind of behavior designed on propose ,or it's a bug?

I first though the upper val(or the val on left), sould also map to the 'left'/'high' part of value.
but it turns out is opposite, is there any reason why design like this?

Dolu1990, 2021-04-21:
Hi,

That's one not so good design in SpinalHDL, the Cat function behave in a not so ortigonal way.

if you Cat(a,b)   then a will go in MSB while B will be LSB (it will match as if you where writing the bits on paper like a word)
if you Cat(List(a,b)) it will do the reverse, as now it is realy a list so first element of the list will go to LSB

So, to me it is the expected behaviour, then about if that behaviour is nice, that's another question ^^

The assign from bits will assign LSB to the first bundle element, MSB to the last bundle element. This is i think resonable. Things are seen as a list of bits (starting from LSB) which will be connected to a list of Bundle attribut (starting from the first Bundle attribut)

ArcheyChen, 2021-04-22:
I understand it now, but when I want to split a value, what is the best way to do so.
like I'm trying to split a [addr] to [tag][set][offset]
I can't use something like `Cat(tag,set,offset) := addr` because it seems like Cat will make a copy of wires, instead of making reference

Dolu1990, 2021-04-22:
Right, currently, there is no nice way of handeling (tag,set,offset) := addr like syntaxes, but that would be usefull.

I'm a bit worying about Cat returning reference instead of copy, as it could be a unexpected behaviour in others situation.
Another solution would be to allow things like Vec(a,b,c).assignFromBits(raw)

As Vec explicitly do reference to the construction arguement list. But so far there is a few things that we would have to patch in the Vec class to allow arguements types to not be all the sames.

Would that make sense ?

volatile-static, 2021-04-23:
> I can't use something like `Cat(tag,set,offset) := addr` because it seems like Cat will make a copy of wires, instead of making reference

I've heard that @jijingg will add this in the future.

ArcheyChen, 2021-04-25:
> Right, currently, there is no nice way of handeling (tag,set,offset) := addr like syntaxes, but that would be usefull.
>
> I'm a bit worying about Cat returning reference instead of copy, as it could [...]

@Dolu1990 

what will happen if Cat returning reference? using a modify version of Vec(a,b,c) can is acceptable,but the problem is Cat is use to  pack something, and Vec is use to generate a list of something. And it will be confusing if one use `raw := Cat(a,b,c)`,but have to use`Vec(a,b,c).assignFromBits (raw)`

As for the Bundle assignFromBits problem, I think we can add a trait like "InOrder" ,then we can use `class myBundle extends Bundle with InOrder`, and with this trait, elements will cat from MSB to LSB(which will make sense a lot), and will not break the  compatibility

Dolu1990, 2021-04-27:
I added ways to assign a set of signal at once : 

https://github.com/SpinalHDL/SpinalHDL/commit/3bab9084945a27d0238ae347ba05c7e1a03f37ad#diff-f3530082c8b1b6e277cf053c28e809197b973eccc0adadcb80178b84eed87bf2R102
=> 
(a,b,c) := B"111100"

> what will happen if Cat returning reference?

All code which assume that cat return a new signal  will fail : 

```scala
val a = Cat(x,y,z) //Create a new signal assigned by x,y,z
when(???){ a := 42 }  //assign the signal a, not x y z
```

> As for the Bundle assignFromBits problem, I think we can add a trait like "InOrder" ,then we can use class myBundle extends Bundle with InOrder, and with this trait, elements will cat from MSB to LSB(which will make sense a lot), and will not break the compatibility

With a trait as a high level API we can, but the low level API should be an attribut in the Bundle class itself i would say.

Dolu1990, 2021-04-27:
> elements will cat from MSB to LSB(which will make sense a lot), and will not break the compatibility

I gess it depend the context. For instance, for C structs, the first element of the struct is placed in the low address (like in SpinalHDL)

If you use the bundle to define a set of different things, SpinalHDL "follow" the C way.
If you use the bundle to define a set of different parts which together form a readable word, then the InOrder make sense i gess

ArcheyChen, 2021-04-28:
> I added ways to assign a set of signal at once :
> 
> [3bab908#diff-f3530082c8b1b6e277cf053c28e809197b973eccc0adadcb80178b84eed87bf2R102](https://github.com/SpinalHDL/SpinalHDL/commit/3bab9084945a27d0238ae347ba05c7e1a03f37ad#diff-f3530082c8b1b6e277cf053c28e809197b973eccc0adadcb80178b84eed87bf2R102)
> =>
> (a,b,c) := B"111100"
> 
> > what will happen if Cat returning reference?
> 
> All code which assume that cat return a new signal will fail :
> 
> ```scala
> val a = Cat(x,y,z) //Create a new signal assigned by x,y,z
> when(???){ a := 42 }  //assign the signal a, not x y z
> ```
> 
> > As for the Bundle assignFromBits problem, I think we can add a trait like "InOrder" ,then we can use class myBundle extends Bundle with InOrder, and with this trait, elements will cat from MSB to LSB(which will make sense a lot), and will not break the compatibility
> 
> With a trait as a high level API we can, but the low level API should be an attribut in the Bundle class itself i would say.

@Dolu1990 

This is remind me an other intresting question ---- it seems like spinal don't have `WireInit` or something like it.

I believe that `:=` is using for assign, and `=` is to get a reference.
So,`val a = Cat(x,y,z)` should be a reference to `x,y,z` ,in stead, it return a copy.

In a lot of case, we want to make a copy of some signals, instead of getting a reference.
In chisel, it seems like it's call `WireInit`.

without that ,I have to write codes like this:
```scala
val wireA = myComplexBundle(myConfigs)
……
val muxVal = myComplexBundle(myConfigs) // or cloneOf(muxVal), but cloneOf need case class, and case class can't inheritance
muxVal := wireA //default val
when(cond){
  muxVal := valB
}
```
with `WireInit` we can do like this 
```scala
val wireA = myComplexBundle(myConfigs)
……
val muxVal = WireInit(wireA)
when(cond){
  muxVal := valB
}
```

by the way, about it's name. I think 'WireCopy' is better than `WireInit`

Dolu1990, 2021-04-28:
> This is remind me an other intresting question ---- it seems like spinal don't have WireInit or something like it.

It has, it is : 

```scala
val miaou = CombInit(xxx) //Comb mean combinatorial
```

> I believe that := is using for assign, and = is to get a reference.
So,val a = Cat(x,y,z) should be a reference to x,y,z ,in stead, it return a copy.

= is a pure scala thing to get the reference to the result of the expression on the left. SpinalHDL actualy never interpret the left operand of := and = differently. Basicaly, there is a API in SpinalHDL to redirect assignement made to a given Data to something else, and that's how (for instance) redirection of  myBits(4) := True works. SpinalHDL see a Bool assigned by another Bool, but will redirect that assignement to the original Bits.

This is currently only used for BitVector sub parts accesses and a few case for Mem to infer write ports. But with that way of doing things, it is either all on or all off for a given operation kind (ex Cat)

Currently, Cat is considered as a expression, not a reference / new data structure. I would say Tuple fit better your the use case ?

> In a lot of case, we want to make a copy of some signals, instead of getting a reference.
In chisel, it seems like it's call WireInit.

Right, they changed that between Chisel 2 and 3. Their type system changed quite a lot.

Basicaly, in SpinalHDL the default when you instanciate something is that this thing is added to the netlist.
If you want to capture a datatype at runtime for later use, there is the :

```scala
val dataType = HardType(Bits(8 bits))
```

In my opinion, the Chisel way is sort of getting away from the internal DSL nature of stuff and try to ""fake"" things as much as possible to make it look like a external DSL. Which has its pro and con.

So

```scala
val wireA = myComplexBundle(myConfigs)
……
val muxVal = CombInit(wireA)
when(cond){
  muxVal := valB
}
```

should be fine in SpinalHDL

> and as for the 'Cat' problem, it should write like this:val a = WireInit(Cat(x,y,z))

What is you opinon on that actual SpinalHDL working syntax (not the same case) : 

```scala
val a = x + y
when(something){
   a := 42
}
```

> by the way, about it's name. I think WireCopy is better than WireInit

Ahhh i guess here the idea was to preserve the same "syntax" than for register, RegInit(U(42, 8 bits))

ArcheyChen, 2021-05-02:
I guess you are right, `=` not always means reference, because we can't assume every function returns a reference or copy
```scala
val a = someFunc(x,y) // Copy or reference? We can't know 
when(something){
   a := 42
}
```
I think what we are really talking about is that, some function in spinal will return a copy while we might need it's reference version.
Like `Cat` is a copy version, and the `(a,b,c)` is a reference version.
both of them will be useful in certain cases.

--------

So, if we thinking in this way
1. `asBits` should have a reference version. Like we want to assign some part of bits of a bundle `myBundle.asBits(10 downto 5)`
2. `a := (x,y,z).asBits` looks odd, because we can assign Bits to it `(a,b,c) := someBits` . Can't we just simply make it default as a Bits,just like`Cat`?  `a := Cat(x,y,z)`
3. is there any other function that will need a reference version?

Dolu1990, 2021-05-17:
Sorry for the late response 

There is maybe some more to think about : 

```scala
scala> x.toInt
res0: Int = 1

scala> var x = 1.0
x: Double = 1.0

scala> x.toInt
res1: Int = 1

scala> x.toInt = 4
<console>:12: error: value toInt_= is not a member of Double
       x.toInt = 4
         ^

``` 

In other words, the actual implementation in SpinalHDL, "follow" the Scala ways "more or less".

1) currently, the api to do this is myBits.assignFromBits(something, h, low). That's quite verbose
2) My guess is that Cat(x,y,z) returning the reference to x,y,z is quite verilog like (in VHDL there is no such thing)
What's about having a := (x,y,z) ? I guess it would be possible with some scala trickery
3) To be honnest, i tried to realy keep them to a minimum, probably to follow the scala nature of cast (see the x,toInt = 4) example

Currently references are only used in :  
- BitVector(x downto x) stuffes
- Vec(x,y,z)(sel) := miaou

What is your opinion ? Does the x.toInt = 4 example  changed something ?