SpinalHDL/SpinalHDL #409: bugs of reassignment

dxsm, 2021-04-21:
[image]
As show in the picture，I reassign myBits_8bit，but generate rtl reassign to myBits_32. Is it a bug?

typingArtist, 2021-04-21:
Your assignment `val myBits_8bits = myBits_32(7 downto 0)` makes `myBits_8bits` an alias for the 8 LSBs of `myBits_32`. Thus assigning to it assigns to `myBits_32`.

dreamflyings, 2021-07-26:
@dxsm try this
`class TestBitsAssignTop(width: BitCount) extends Component{`
`  val myBits_32 = B(32 bits, default -> True)              `
`  val myBits_8bits = Bits(8 bits)                          `
`  myBits_8bits :=  myBits_32( 7 downto 0)                  `
`  val myBits = Reg(Bits(32 bits))                          `
`                                                           `
`  myBits := myBits_32                                      `
`  myBits_8bits(3 downto 0) := B"4'b0010"                   `
`}                                                          `
`                                                           `
`object Main extends App{                                   `
`  SpinalVerilog(new TestBitsAssignTop(32 bits))            `
`}                                                          `

Tectu, 2026-06-01:
Closing this due to inactivity.