SpinalHDL/SpinalHDL #411: Is there any syntax  to choose some signals in a bus, like "except" or "payload"?

ArcheyChen, 2021-05-02:
due to some reasons ,sometimes I want to create my own bus(not using Stream or Valid)
like:
```scala
class MyBus extends Bundle with IMasterSlave {
  val ready = Bool()
  val valid = Bool()
  val data1 = UInt(32 bits)
  val data2 = UInt(32 bits)

  override def asMaster()={
    out(valid,data1,data2)
    in(ready)
  }
  def fire = ready && valid
}
```
And I want to add a buffer to it, like:

```scala
class MyTopLevel extends Component {
  val io = new Bundle {
    val upper = slave(new MyBus)
    val lower = master(new MyBus)
  }

  val buffer = Reg(io.upper)
  val bufferValid = RegInit(False)

  io.upper.ready := !bufferValid
  io.lower.assignSomeByName(buffer)
  io.lower.valid := bufferValid
  io.lower.ready.assignDontCare()

  when(io.upper.fire){
    buffer := io.upper
    bufferValid := True
  }
  when(io.lower.fire){
    bufferValid := False
  }
}
```

but the code above will report `HIERARCHY VIOLATION`, and, it will buffer the unneccsery `valid` and `ready`

It's intresting that when I using `Stream` or `Valid`,I can use both `bus.payload.mysignal` and `bus.mysignal` to access mysignal
It looks like `payload` is warping the payload, but somehow I can still access then directory.I don't know how to do that.

also, in some cases,a Bundle might mix signals from diffrent Modules,some times we will have to filter some signals to operate instead of operating all signals, so I think some syntax like `except` , `choose`, `getOutput`,`getInput` will be useful
```scala
// Some parts should use assignSomeByName or assignAllByName, but I just use := for show case
//except
io.myBus := myBundle.except(signal1,signal2)
io.myBus.except(signal1,signal2) := myBundle
//choose
io.myBus := myBundle.choose(data1,data2)
io.myBus.choose(data1,data2) := myBundle.choose
//getIn/Output
myBuf := io.myBus1.getInput
io.myBus2.getOutpt := myBuf
```

So
1. how to warp some signals like the way `Stream` and `Valid` does?
2. I think some filter syntax will be useful

Dolu1990, 2021-05-03:
> but the code above will report HIERARCHY VIOLATION

From what is it ?

> It's intresting that when I using Stream or Valid,I can use both bus.payload.mysignal and bus.mysignal to access mysignal
It looks like payload is warping the payload, but somehow I can still access then directory.I don't know how to do that.

That's a trick that i added that way : 

https://github.com/SpinalHDL/SpinalHDL/blob/b42ad071474c2042bf13d9ff30147b0e378b322d/lib/src/main/scala/spinal/lib/Utils.scala#L692

> except  choose

what are those two primitives exactly do ?

ArcheyChen, 2021-05-03:
> > but the code above will report HIERARCHY VIOLATION
> 
> From what is it ?
> 

Because the code ` buffer := io.upper` will buffer all signals ---- including `ready`
when I do `io.lower.assignSomeByName(buffer) ` , it will try to assign `io.lower.ready` ---- which is a input signal, and that will cause `HIERARCHY VIOLATION`

and  I don't know how to 'unassign' it 

> 
> > except  choose
> 
> what are those two primitives exactly do ?

like if I have a bundle, and it's data comes from two sources
```scala
class MyBundle extends Bundle{
  val a = UInt(32 bits)
  val b = UInt(32 bits)//from source A
  
  val c = UInt(32 bits)
  val d = UInt(32 bits)//from source B
}
class MyComponent extends Component{
  val finalData = new MyBundle
  val sourceA = new MyBundle
  val sourceB = new MyBundle
  
  finalData.a := sourceA.a
  finalData.b := sourceA.b
  
  finalData.c := sourceB.c
  finalData.d := sourceB.d 
}
```
and if there are multiple sources I will have to connect them wire by wire.
if with `except` or `choose` , we can do like this:

```scala
finalData.choose(a,b).assignAllByName(sourceA) // only connect a,b in finalData
finalData.assignSomeByName(sourceB.choose(c,d)) // only use c,d in sourceB to connect to finalData
//// or ,we can use the except way

finalData.except(c,d).assignAllByName(sourceA) // assign everything in finalData except a,b
finalData.assignSomeByName(sourceB.except(a,b)) // use everything in sourceB except a,b, to assign finalData

```

I think they will be useful in some cases, don't they?

Dolu1990, 2021-05-03:
> Because the code buffer := io.upper will buffer all signals ---- including ready

Ahhh right, that's why in Stream/Flow it explicitly group things into the payload Bundle

> except or choose

I see now, so basicaly, i'm trying to avoid falling in those usecase, and so far if i have to go in those kind of case, i do a 
finalData.assignAllByName(sourceA)
finalData.valid.removeAssignements()
finalData.ready.removeAssignements()

But that's tricky.

Having a choose(a,b).assignAllByName working maybe tricky, maybe the best would be to combine those function : 

finalData.assignAllExcept(sourceA, finalData.valid, finalData.ready) ?

ArcheyChen, 2021-05-04:
> Having a choose(a,b).assignAllByName working maybe tricky, maybe the best would be to combine those function :
> 
> finalData.assignAllExcept(sourceA, finalData.valid, finalData.ready) ?

it will work fine when we just want to assign a bundle.
but I'm thinking,  will this work on a buffer or fifo?

you see, `Reg(complexBundle)` will make a Reg that contain all signals in `complexBundle`, some of which we are not going to use.
when the code is complex, I can't konw if the EDA will delete those useless signals.
what do you think inthis kind of situation?

Dolu1990, 2021-05-04:
>  I can't konw if the EDA will delete those useless signals. what do you think inthis kind of situation?

Yes, EDA tools will remove useless stuff, and i often rely on that ^^