SpinalHDL/SpinalHDL #413: Bug(?) report: using `:=` and `readSync` at the same time will generate 2 Mems

ArcheyChen, 2021-05-03:
```scala

class MyTopLevel extends Component {
  val we = Bool()
  val addr = UInt(10 bits)
  val mem = Mem(UInt(32 bits),1024)
  mem.generateAsBlackBox()
  we := True
  addr := 10
  val readData = mem.readSync(addr)
  when(we){
    mem(addr) := 123
  }
}
```

this code will generate 2 Mem: a sync one and a async one.

this might cause by using readsync and async write at the same time.

Dolu1990, 2021-05-03:
This is very likely due to mem(addr) := 123   which internaly in SpinalHDL already prepear a async read in case of you want to read the mem(addr)  statement. That's a bit dump.

Can you give a try using the explicit mem.write statement instead of mem(addr) := 1 ?

normaly that one whould be fine.

ArcheyChen, 2021-05-04:
> This is very likely due to mem(addr) := 123 which internaly in SpinalHDL already prepear a async read in case of you want to read the mem(addr) statement. That's a bit dump.
> 
> Can you give a try using the explicit mem.write statement instead of mem(addr) := 1 ?
> 
> normaly that one whould be fine.

Yes, using `mem.write` with `mem.readSync` will fix this problem

but what I mean is that, we should add some codes to check if one using `:=` and `readSync` at the sametime.we should not allow them to use in the same Mem

Dolu1990, 2021-05-04:
> but what I mean is that, we should add some codes to check if one using := and readSync at the sametime.we should not allow them to use in the same Mem

To be honnest it hink  the fancy mem(addr) := 123  syntax could be depecated as it can easily be missued.

Implicit memory port inferation via a proper function is my favorit so far.

> := and readSync at the sametime

It can make sense to be used at the same time, basicaly, the issue you had is that SpinalHDL didn't removed some useless internals early enough for the blackboxification to not be confused