SpinalHDL/SpinalHDL #414: GHDL simulation v1.0 regression

jiegec, 2021-05-04:
As discussed in https://github.com/SpinalHDL/SpinalHDL/pull/412#issuecomment-831882384, there is a regression in simulation:

Run `testOnly spinal.tester.scalatest.Axi4CrossbarTesterCocotbBoot` in sbt:

- GHDL 0.36-dev (Debian 0.35+git20181129+dfsg-3) [Dunoon edition]: Completes 
- GHDL 1.0.0 (Debian 1.0.0+dfsg-1) [Dunoon edition]: Never completes

Yuwenfeng2019, 2021-05-04:
Thanks!
I used the Fedora Spin 33 XFCE on my Raspberry Pi, and is using the GHDL 1.0dev Dunnon edition:
[image]
It seems that the blocked cases may caused by the latest GHDL version, but as GHDL is mainly written in Ada, it is difficult to debug.
If these issues are really caused by GHDL version, then the test result on X86 should have the similar phenomenon when testing with GHDL 1.0?

Dolu1990, 2021-05-12:
@SomeParticles Hi ^^
This issue may concern you.

bellaz89, 2021-05-12:
Hi, could you also verify if this behavior is present with all the three GHDL backend? (mcode, LLVM, GCC) Thanks!

bellaz89, 2021-05-12:
@Dolu1990 Are we sure that this is a problem of the SpinalHDL GHDL backend? 
Doesn't  spinal.tester.scalatest.Axi4CrossbarTesterCocotbBoot use Cocotb? In this case that would mean that there is some problem with Cocotb GHDL backend.

Dolu1990, 2021-05-12:
@bellaz89 @SomeParticles 
I'm sorry, i was reading it as "there is a issue in the SpinalHDL based simulation" :/

> Doesn't spinal.tester.scalatest.Axi4CrossbarTesterCocotbBoot use Cocotb?

Right

> In this case that would mean that there is some problem with Cocotb GHDL backend.

Right, and it already happend by the past, that's why i realy tried to fix versions to working one in travis to avoid wasting people time on transiants.

So i would say, let's wait a bit.

typingArtist, 2021-06-21:
This problem exists on Fedora 34 as well.
```
$ ghdl --version
GHDL 1.0-dev () [Dunoon edition]
 Compiled with GNAT Version: 11.0.0 20210210 (Red Hat 11.0.0-
 GCC back-end code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2020 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

Dolu1990, 2021-06-22:
So, a easy way to reproduce the issue is to run StreamTester2CocotbBoot

basicaly, there, the VHDL statement "fifoAPop_valid <= fifoA_io_pop_valid;" isn't properly applied by the simulation and instead, fifoAPop_valid is set to zero all the time while fifoA_io_pop_valid is having some fun XD

Not sure if that's a pure GHDL issue, or if things get corrupted via cocotb / VPI / GHDL

typingArtist, 2021-06-23:
Found the problem. See SpinalHDL/cocotblib#10. It seems that GHDL 1.0 takes driving an output from the outside serious, so when you drive it LOW in the Cocotb monitor and then ask for the value of the output signal, you actually get a static LOW indication. Previously, you still got the DUT output signal.

typingArtist, 2021-06-24:
@jiegec  I think that with the latest fix and merge, the regression is gone and we can close the issue.

closed 2021-06-24 by Dolu1990