SpinalHDL/SpinalHDL #421: When does spinal.sim support vcs or modelsim

qgzln, 2021-05-20:
I am FPGA developer，I often use some ip core，so，When does spinal.sim support vcs or modelsim

Dolu1990, 2021-05-24:
So basicaly, it would require implementing a vcs + a modelsim backend, as it was done for https://github.com/SpinalHDL/SpinalHDL/blob/dev/sim/src/main/scala/spinal/sim/VpiBackend.scala#L330

I do not have the availability nor the access to vcs/modelsim, so, currently it isn't planned.

dreamflyings, 2022-04-12:
> So basicaly, it would require implementing a vcs + a modelsim backend, as it was done for https://github.com/SpinalHDL/SpinalHDL/blob/dev/sim/src/main/scala/spinal/sim/VpiBackend.scala#L330
> 
> I do not have the availability nor the access to vcs/modelsim, so, currently it isn't planned.

Thank you very much for your detailed explanation and optimization of this.

Readon, 2023-01-04:
fixed by #604 #644 #650 #651.

closed 2023-01-04 by Readon

numero-744, 2023-01-04:
@Readon Do we declare it stable in RTD?

Readon, 2023-01-04:
> @Readon Do we declare it stable in RTD?

I could not say about that, because I am not using a commercial tools now. 
Someone reported that is working, some are still meet a problem.
I think we need a brief tutorial on how to make it work by someone using it.

Dolu1990, 2023-01-05:
> I could not say about that, because I am not using a commercial tools now.

Same

numero-744, 2023-01-05:
Might sound crazy: has anyone considered contacting Synopsys to know if they would be interested in testing that interface, or participate in its implementation by any way?

Said another way; to ask them if they are "interested into bringing to their tools better support for a language that let their clients improve their productivity, so that both Synopsys and their clients can be more competitive".

Dolu1990, 2023-01-06:
> Might sound crazy: has anyone considered contacting Synopsys to know if they would be interested in testing that interface, or participate in its implementation by any way?

There is not much money for them to make there, so no incentives.

> interested into bringing to their tools better support for a language that let their clients improve their productivity

That's one perspective, but i don't think they could share it ^^