SpinalHDL/SpinalHDL #422: support Rounding for UFix/SFix

xiejin03, 2021-05-21:
Hello, 
I tried the UFix and SFix, but found it only support truncation between different resolutions.
Could SpinalHDL support rounding between different resolution fixed point values?

Dolu1990, 2021-05-24:
Hi,

Sorry, i'm not sure to exactly understand, can you give a practical example to help ?

xiejin03, 2021-05-25:
Hi, thanks for the reply. Here is an example:   

    val inp_pix = SFix(8 exp, -8 exp)
    val coef = SFix(4 exp, -4 exp)
    val compute_pix = inp_pix * coef    // multiply the input pixel with coef, and the compute_pix has 12 bit fraction.
   
    val cur_pix = SFix(8 exp, -8 exp)
    // I want to clip the compute_pixel into SFix(8, -8), but currently only truncated is available.
    // round 12bit fraction into 8 bit fraction, and saturate the 12 bit integer into 8 bit integer. 
    cur_pix := compute_pix.truncated

    clip SFix(12 exp, -12 exp) into SFix(8 exp, -8 exp) is very common because our image pipeline keeps pixel signal width unchanged for module input/output. and truncated is not enough because algo team uses clip.

xiejin03, 2021-05-25:
Thanks @jijingg . I know that SInt and UInt supports rounding, but what I need is SFix() and UFix() to support rounding. And SFix() and UFix() is very useful for image processing, because SInt and UInt is not straight-forward enough.

From your answer, it seems there is a SQ data type. but I didn't see it in SpinalHDL document.
And SFix(8 exp, -8 exp) is a 16 bit int, whose high 8 bit is signed integer part and low 8 bit is fraction part. Seems Sint() and SQ() has different meaning of the first argument.

From a RTL owner view, I prefer SInt and Uint because it is more straightforward. we don't have to calcualate how many bits for the integer part. And Sint already well documented. why not merge the SQ into Sint and support rounding in Sint?

xiejin03, 2021-05-25:
@Dolu1990 sorry I have to use Chinese to convince @jijingg that SFix() and UFix() is very useful in ISP.
@jijingg 下面是海思3559 的芯片手册里面，关于定点数的说明，在ISP 领域，SFix() 和 UFix() 是最适合描述像素的一种方式。所以强烈建议支持。

本文中ISP采用U*.*、S*.*表示无符号数和有符号数。如：U8.8表示数据类型为无符号数，整
数部分8bit，小数部分8bit；同理，S8.8表示数据类型为有符号数，整数部分8bit（包括1bit符
号位），小数部分8bit。

jijingg, 2021-05-25:
> SFix() and UFix() is very useful for image processing, because SInt and UInt is not straight-forward enough.

I'm not sure if it's necessary to specifically abstract a data type of SFix UFix, There are two difficulties
1.  Too much flexible and high level 
for example :
two 8 bit adder will lead different width. Some people may think it's convenient, but I think it will lead to complexity and unpredictability, it's diffcult for debug. the fix operation should explicit show on dagram. not give the tool auto infer。 
```
S8.0 + S8.0 = S9.0
S8.7 + S8.7 = S9.7
S8.7 + S8.0 = S16.7
S8.4 + S8.0 = S13.4
```
2. How to design Api operation with SInt  UInt
if we need force all SInt UInt to SFix, UFix， it seems give a new high level or new luaguage .
```
val a = SFix(8,7)
val b = UInt(8 bit)
val C = a + b    // what width should C be? 
val D = a * b    // what FixPoint showd D be? 
```
i have no clear idea, if you have pretty solution, show that .

I aggree add SQ/UQ tag on SInt/UInt not more straight-foward , but  This is a more realistic solution， not only ISP ,and in communication basband process , the SQ/UQ are very common

jijingg, 2021-05-25:
@xiejin03  the second question about QFormat expression .  there have not standard for expression .
different team or company have different expression .It can only be said that habits are different.
On the three situations in this discussion ：
1. SFIX(8, -8)  means 8 - (-8) = 16bit width 
2. S8.7 means 8bit width , 7 bit integer, 0 bit fraction , 1bit sign bit
3. SQ(8,7) means 8 bit width ,7 bit fraction 0 bit integer , 1bit sign bit

but in my view,  the expression should show 3 thing directly

**1. is signd?
2. total width 
3. fraction postion** 

so i use `SQ(8,7)`  `S` means Signed ,first `8` means `8` bit width ,and second `7 `means `7` bit fraction 

It may be different from the your favorite way of expression, but as long as the agreement is unified it's not big deal.

if you have a better suggestions that can be discussed

jijingg, 2021-05-25:
> 本文中ISP采用U*._、S_.*表示无符号数和有符号数。如：U8.8表示数据类型为无符号数，整
> 数部分8bit，小数部分8bit；同理，S8.8表示数据类型为有符号数，整数部分8bit（包括1bit符
> 号位），小数部分8bit。

最好还是用英文，虽然我英文也不咋地，希望讨论可以被其他人检索参与， 你可以加我微信htgoco讨论

xiejin03, 2021-05-26:
My idea:
1. UInt/SInt is different data type from SFix/UFix. We don't need to support the operation for UInt/SInt + SFix/UFix.
2. SFix(m, n) and SQ(m+n+1,  n) actually mean the same thing. UFix(m, n) and UQ(m+n, n) means the same thing too. Could we do something like "alias" to solve this problem?
3. for different adder width, SInt/UInt already has +, +^, +| operators. Could we do the same thing for SFix/UFix/SQ/UQ?

closed 2021-06-02 by xiejin03

Dolu1990, 2021-06-02:
@xiejin03 

> for different adder width, SInt/UInt already has +, +^, +| operators. Could we do the same thing for SFix/UFix/SQ/UQ?

Sure we can

> UInt/SInt is different data type from SFix/UFix. We don't need to support the operation for UInt/SInt + SFix/UFix.

Yes i agree, but in the same time, different people will prefer different ways of doing things as they have different background and problems to solve. As SpinalHDL came mostly from experiementations, i do not want to limite the ways of doing things, but keep ways open.

Ultimately, my ideal abstract type would be one which can be used in place of UInt, SInt, UFix, SFix, all at once. The main issue is that i just do not have any proper application to use it myself, so i don't want to start myself something i can't properly manage, but if somebody does, it is good ^^