SpinalHDL/SpinalHDL #425: Bit extraction from Vec

huang-jl, 2021-05-23:
Hi, I meet a problem with bit extraction from signal inside a `Vec`

```scala
val data = Vec(Bits(32 bits), 8)
val addr = UInt(3 bits)
//...
data(addr)(0, 8 bits) := U(0, 8 bits)
```

The error after running is:
```
WIDTH MISMATCH on (toplevel/data_0 :  Bits[32 bits]) := (B"00000000" 8 bits) at 
...
WIDTH MISMATCH on (toplevel/data_7 :  Bits[32 bits]) := (B"00000000" 8 bits) at 
```

How can I assign only some bits of one signal inside `Vec` (which is `data` in the example)?

huang-jl, 2021-05-23:
@Dolu1990 Looking forward to your help.

Ncerzzk, 2021-12-17:
I also met the problem.  Do you solve it? @huang-jl 

@Dolu1990

Dolu1990, 2021-12-17:
Ahh damned, i missed that issue.
I will give some checks, i think i may have fix something related to that. but i have to check.

Dolu1990, 2021-12-17:
Else there is still the work around to manualy handle it : 
```scala
for(i <- 0 to 8) when(addr === i) { data(i)(0, 8 bits := U(0, 8 bits) }
````

Dolu1990, 2021-12-18:
I just tested, it was fixed. 1.6.1 should work.

closed 2021-12-18 by Dolu1990