SpinalHDL/SpinalHDL #429: Question about `Mem.readSync()` usage.

selendym, 2021-05-26:
Hello.

I'd like to ask about `Mem.readSync()` usage. Consider the following example:
```scala
package test

import spinal.core._

class Test[ T <: Data ]( val dataType: HardType[ T ], val capacity: Int ) extends Component {
    val io = new Bundle {
        val readEnable = in( Bool() )
        val readAddr = in( UInt( log2Up( capacity ) bits ) )
        val readData0 = out( dataType() )
        val readData1 = out( dataType() )
        val readData2 = out( dataType() )
    }

    val mem = Mem( dataType, capacity )

    // NOTE: This works.
    io.readData0 := mem.readSync( io.readAddr, io.readEnable )

    // NOTE: This doesn't work and fails with: "LATCH DETECTED from the combinatorial signal (toplevel/io_readData1 : out Bits[32 bits])".
    when( io.readEnable ) {
        io.readData1 := mem.readSync( io.readAddr )
    }

    // NOTE: This works.
    val doReadData = False
    io.readData2 := mem.readSync( io.readAddr, doReadData )
    when( io.readEnable ) {
        doReadData := True
    }
}

object Generate extends App {
    SpinalConfig( netlistFileName = "Test.vhdl" ).generateVhdl( new Test( Bits( 32 bits ), 42 ) ).printPruned()
}
```
I'm puzzled by the failure of the second form (`readData1`), since it should (should it?) in practice be equivalent to the other two.
The second form is arguably the clearest when there's a lot of nesting and I'd like to make it work somehow, so is there some good or idiomatic way around this error?

Best regards.

Dolu1990, 2021-05-27:
io.readData1 is a combinatorial signal, in the second form you conditionaly assign it => latch.
In general i would say, do not use the Mem API to infer stuff in a VHDL/Verilog way, but instead, see the Mem API as a way to instanciate memory ports (not inferring memory port)

Does that make sense ?

selendym, 2021-05-27:
Indeed, the above example assumes that `mem` is inferred with an implicit output reg and no read latency, matching simulation output.

Would it make more sense to consider all memory elements blackboxes, then, or is there any good rules of thumb when to blackbox and when not to (say, expected size, etc.)?

closed 2021-05-28 by selendym