SpinalHDL/SpinalHDL #430: How can I dynamically generate module port using SpinalHDL?

xiejin03, 2021-05-27:
I want to use SpinalHDL to generate a NIC, and I have a config file which indicates how many masters and slaves the NIC has.
The top module's input and output port could be AXI or AHB or APB, which is also decided by the config file, So I cannot use parameter to solve the problem.
I don't know if there is any way to dynamically generate the module in SpinalHDL.
If there is, could anyone tell me how? Thanks a lot!

Dolu1990, 2021-05-29:
For instance, the StreamFifo, it has types parameters using   [T <: Data](dataType : HardType[T], ...)

That's one way of  having types parameters in a staticaly typed language. 
Else you can have to do some stuff like puting the reference to the bus somewhere but not puting the exact scala type.

Also, some way of doing this kind of things could be with HashMap to store stuff via some keys (identifier) and retreiving the actual implemented value.

xiejin03, 2021-06-02:
Thanks @Dolu1990 , I will close this issue.

closed 2021-06-02 by xiejin03