SpinalHDL/SpinalHDL #431: Simulated signal values don't change on active clock edge, only in the waveform

rpls, 2021-05-30:
I'm having some trouble with the simulator, especially with the behavior of `waitSampling{,Where}`. It seems that the registered output signals of my components, only change after very slightly after the active edge of the clock. For example, when I use `      waitSamplingWhere(dut.io.ready.toBoolean)` to wait for a `ready` signal, the function will only return after `ready` has been high for one cycle. Same with `waitSampling`; if I have a signal that i expect to change after one cycle it doesn't appear to be changed. The generated waveform appears to be perfectly fine and exhibits the right behavior. I wonder if I misunderstood the meaning of `waitSampling{,Where}`. Below is a minimal example. It's a simple component that essentially contains a shift register that one can cycle data into, do some work on it, and optionally cycle data out. A minimal fix is a `sleep(1)` right after a `waitSampling`, then the signals appear to be updated.

```scala
class SomeComponent extends Component {
  val io = new Bundle {
    val start = in Bool
    val ready = out Bool
    val cycle = in Bool
    val data_in = in Bits(4 bit)
    val data_out = out Bits(4 bit)
  }

  val state = Vec(Reg(Bits(4 bit)), 16)

  io.ready := False
  io.data_out := state(0)

  val fsm = new StateMachine {
    val CYCLE: State = new State with EntryPoint {
      whenIsActive {
        io.ready := True
        when(io.cycle) {
          state.zip(state.slice(1, state.length)).foreach{case (a, b) => a := b}
          state.last := io.data_in
        }
        when(io.start) {
          goto(DOSTUFF)
        }
      }
    }
    val DOSTUFF: State = new StateDelay(cyclesCount=16) {
      whenCompleted {
        goto(CYCLE)
      }
    }
  }
}

class LateCycleTest extends AnyFunSuite {
  test("cycle comes late") {
    val compiled = SimConfig.withFstWave.compile(new SomeComponent)
    compiled.doSim { dut =>
      dut.io.start #= false
      dut.io.data_in #= 0
      dut.io.cycle #= false
      dut.clockDomain.forkStimulus((16 MHz).toLong / 1e6.toLong)
      dut.clockDomain.waitSamplingWhere(dut.io.ready.toBoolean)
      Seq.range(0, 16).foreach { i =>
        dut.io.data_in #= i
        dut.io.cycle #= true
        dut.clockDomain.waitSampling()
        dut.io.cycle #= false
      }
      /* Problem 1: start held high for 2 cycles instead of 1 */
      dut.io.start #= true
      dut.clockDomain.waitSamplingWhere(!dut.io.ready.toBoolean)
      dut.io.start #= false
      /* Problem 2: Waits until ready has already been high for 1 cycle */
      dut.clockDomain.waitSamplingWhere(dut.io.ready.toBoolean)
      Seq.range(1, 17).foreach { i =>
        dut.io.cycle #= true
        print(f"data_out is ${dut.io.data_out.toInt} before, ")
        dut.clockDomain.waitSampling()
        /* Problem 3: data_out hasn't changed (although it did in the Waveform) */
        /* Fix is: sleep(1) right afterwards, to give the sim time to change */
        println(f"${dut.io.data_out.toInt} after waitSampling")
        dut.io.cycle #= false
      }
      dut.clockDomain.waitSampling(2)
    }
  }
}
```

Dolu1990, 2021-05-30:
Hi,

> It seems that the registered output signals of my components, only change after very slightly after the active edge of the clock.

Yes this is the expected behaviour.

Basicaly, the waitSampling() wakeup after the delta cycle the clock  was assigned high, but before the registers had time to propagate their values. Like in VHDL/Verilog testbench.

> A minimal fix is a sleep(1)

This is a way to "trick it", but normaly there is always ways to not having to do that.

Does that make sense ?

rpls, 2021-05-30:
> This is a way to "trick it", but normaly there is always ways to not having to do that.
> 
> Does that make sense ?

I guess it does, if `waitSampling()` is only used to update inputs into components. What would be a nice pattern to to avoid a `sleep(1)`? Say, for example, a component has a `Flow` as an output, and the output payload and/or valid is registered (or derived from a register). The simulation has to monitor the `valid` signal, and if its high sample the output value and wait one clock cycle. The `Flow` potentially outputs up to one value per cycle.

```scala
while(true) {
    waitUntil(flow.valid.toBoolean) /* Can't use waitSamplingWhere here, as it might skip the first high value */
    flow.payload /* Use the payload for something, e.g., check correctness. */
    waitSampling() /* Wait at least one cycle */
    sleep(1) /* How to avoid this? */
}
```

Dolu1990, 2021-05-30:
So, in general, for Flow and Stream i use the StreamMonitor and FlowMonitor which give a callback on each transaction.

But else : 

```scala
while(true) {
  dut.clockDomain.waitSampling()
  if(flow.valid.toBoolean){
    flow.payload /* Use the payload for something, e.g., check correctness. */
  }
}
```

or

```scala
while(true) {
  dut.clockDomain.waitSamplingWhere(flow.valid.toBoolean)
  flow.payload /* Use the payload for something, e.g., check correctness. */
}
```

In other words, do not wait in a "combinatorial manner" ( waitUntil(flow.valid.toBoolean) )

Use only clocked waits should be fine, normaly ?

rpls, 2021-06-15:
I figured it out for my purposes... I guess I tried to implement a "combinatorical" reaction by the simulator (i.e. pull something high if something else is high), but this wasn't the right approach. The approach of having the `ready` (or whatever signal is acknowledgement of the specific bus) driven high all the time by the simulation was better. Thanks!

closed 2021-06-15 by rpls