SpinalHDL/SpinalHDL #439: [Bug Report] Unused signal prune does not work well

zhutmost, 2021-06-05:
Two problems:

1. Declared but unused signal still appear in the generated Verilog. (For eg., variable `b` in the below case.)
2. The builtin Counter Component will generate lots of unused signals. These signals should be pruned.  (For eg., variable `builtinCounter` in the below case.)

```scala
package test

import spinal.core._
import spinal.lib._

class MyNice extends Component {
  val io = new Bundle {
    val clockDividerTick = in Bool()
    val start = in Bool()
    val outVal1 = out UInt(2 bits)
    val outVal2 = out UInt(2 bits)
  }

  val b = Bits(3 bits) // This signal is never used in this module, and should be pruned.

  // The below two counters should have similar outputs in the generated RTL.
  val builtinCounter = Counter(2 bits, inc = io.start)
  val tickCounter = new Area {
    val value = Reg(UInt(2 bits))
    def reset(): Unit = value := 0

    when(io.clockDividerTick) {
      value := value + 1
    }
  }

  when(io.start) {
    builtinCounter.clear()
    tickCounter.reset()
  }

  io.outVal1 := tickCounter.value
  io.outVal2 := builtinCounter.value
}

//Define a custom SpinalHDL configuration with synchronous reset instead of the default asynchronous one. This configuration can be resued everywhere
object MySpinalConfig extends SpinalConfig(defaultConfigForClockDomains = ClockDomainConfig(resetKind = SYNC))

//Generate the MyTopLevel's Verilog using the above custom configuration.
object TestRtlEmitter extends App {
  MySpinalConfig.generateVerilog(new MyNice).printPruned()
}
```
It will generate:
```Verilog 
// Generator : SpinalHDL v1.4.3    git head : 4dd3b62906925bc269aee976e75f8918d4132efb
// Component : MyNice
// Git hash  : 857d51f2a82faeeb9698523c34e86fbc223bbe81

module MyNice (
  input               io_clockDividerTick,
  input               io_start,
  output     [1:0]    io_outVal1,
  output     [1:0]    io_outVal2,
  input               clk,
  input               reset
);
  wire       [0:0]    _zz_1;
  wire       [1:0]    _zz_2;
  wire       [2:0]    b; // <---------------- Here, this statement should not appear.
  reg                 builtinCounter_willIncrement; // <--------- These signals are unused.
  reg                 builtinCounter_willClear;
  reg        [1:0]    builtinCounter_valueNext;
  reg        [1:0]    builtinCounter_value;
  wire                builtinCounter_willOverflowIfInc;
  wire                builtinCounter_willOverflow;
  reg        [1:0]    tickCounter_value;

  assign _zz_1 = builtinCounter_willIncrement;
  assign _zz_2 = {1'd0, _zz_1};
  always @ (*) begin
    builtinCounter_willIncrement = 1'b0;
    if(io_start)begin
      builtinCounter_willIncrement = 1'b1;
    end
  end

  always @ (*) begin
    builtinCounter_willClear = 1'b0;
    if(io_start)begin
      builtinCounter_willClear = 1'b1;
    end
  end

  assign builtinCounter_willOverflowIfInc = (builtinCounter_value == 2'b11);
  assign builtinCounter_willOverflow = (builtinCounter_willOverflowIfInc && builtinCounter_willIncrement);
  always @ (*) begin
    builtinCounter_valueNext = (builtinCounter_value + _zz_2);
    if(builtinCounter_willClear)begin
      builtinCounter_valueNext = 2'b00;
    end
  end

  assign io_outVal1 = tickCounter_value;
  assign io_outVal2 = builtinCounter_value;
  always @ (posedge clk) begin
    if(reset) begin
      builtinCounter_value <= 2'b00;
    end else begin
      builtinCounter_value <= builtinCounter_valueNext;
    end
  end

  always @ (posedge clk) begin
    if(io_clockDividerTick)begin
      tickCounter_value <= (tickCounter_value + 2'b01);
    end
    if(io_start)begin
      tickCounter_value <= 2'b00;
    end
  end

endmodule
```

selendym, 2021-06-07:
Not sure if this was intended or not, but the example code seems to both increment and reset `builtinCounter` with `io.start`.
Perhaps you meant:
```scala
val builtinCounter = Counter(2 bits, inc = io.clockDividerTick)
```

zhutmost, 2021-06-07:
> Not sure if this was intended or not, but the example code seems to both increment and reset `builtinCounter` with `io.start`.
> Perhaps you meant:
> 
> ```scala
> val builtinCounter = Counter(2 bits, inc = io.clockDividerTick)
> ```
> 
> (This change doesn't seem to affect the generated code all that much, though, and just changes the relevant conditions.)

Oh you are right, thx. I have tried, but it doesn't affect the generated RTL very much. The problem still exists.

Dolu1990, 2021-06-07:
>   wire       [2:0]    b; // <---------------- Here, this statement should not appear.
  reg                 builtinCounter_willIncrement; // <--------- These signals are unused.

This is done on purpose, it avoid people being suprise that stuff disapeared. Especialy for builtinCounter_willIncrement which could be used as a debug signal for simulation.
 
By the past, SpinalHDL was realy prunning everything not usefull for the toplevel outputs, but it can lead to very bad cases where   half of the design is pruned because a few wire where not connected XD

zhutmost, 2021-06-07:
@Dolu1990  Thx. Maybe we need an option like `pruneUnusedLogic`, which can be enabled manually. 
Unused signals in the generated RTL may confuse the ASIC verification tools.

zhutmost, 2021-06-07:
So actually the Pruning feature is disabled by default in the latest version? (Even though it have printed something in the log?

Dolu1990, 2021-06-07:
> Thx. Maybe we need an option like pruneUnusedLogic, which can be enabled manually.
Unused signals in the generated RTL may confuse the ASIC verification tools.

Sure it could be added. There is already quite some information in the netlist which indicate which signal are usefull or not.
(BaseType.isVital)

> So actually the Pruning feature is disabled by default in the latest version? (Even though it have printed something in the log?

The feature was the result of how SpinalHDL worked by design by the past, it isn't backed in now.

zhutmost, 2021-06-07:
OK. understood. (We really expect this feature.

Dolu1990, 2021-06-10:
I'm working on it. Got something to work, just have to run a few more tests ^^

Dolu1990, 2021-06-10:
Implemented in https://github.com/SpinalHDL/SpinalHDL/commit/ba6873e8b3cd9b44893412574eb58b6d6164cd90

Dolu1990, 2021-06-10:
If you have any issue with the feature let's me know ^^

closed 2021-06-10 by Dolu1990

zhutmost, 2021-06-11:
OK. Thanks a lot. Let me have a look.