SpinalHDL/SpinalHDL #442: Verilog memory emitter with initial values handles negative values incorrectly.

selendym, 2021-06-09:
Hello.

The Verilog memory emitter with initial values seems to handle negative values incorrectly, while the VHDL emitter doesn't seem to have this problem.

Consider the following example and the generated netlists:
```scala
package test

import spinal.core._

class Test extends Component {
    val mem = Mem( Bits( 32 bits ), 3 )
    mem.initBigInt( List(
        BigInt( 1 ),
        BigInt( 0 ),
        BigInt( -1 )
    ) )
}

object Generate extends App {
    SpinalConfig( netlistFileName = "Test.vhdl" ).generateVhdl( new Test )
    SpinalConfig( netlistFileName = "Test.v" ).generateVerilog( new Test )
}
```

Generated VHDL:
```vhdl
  type mem_type is array (0 to 2) of std_logic_vector(31 downto 0);
  signal mem : mem_type := (
     "00000000000000000000000000000001","00000000000000000000000000000000","11111111111111111111111111111111");
```

Generated Verilog bin file:
```
00000000000000000000000000000001
00000000000000000000000000000000
000000000000000000000000000000-1
```

The problem is the last value in the Verilog bin file, which is not even valid syntax.

Unless I'm mistaken, the problem is likely somewhere around:
- https://github.com/SpinalHDL/SpinalHDL/blob/b6368aaed142fccf51b1b0fea777eb3f911d40da/core/src/main/scala/spinal/core/internals/ComponentEmitterVerilog.scala#L937
- https://github.com/SpinalHDL/SpinalHDL/blob/b6368aaed142fccf51b1b0fea777eb3f911d40da/core/src/main/scala/spinal/core/internals/ComponentEmitterVerilog.scala#L986
- https://github.com/SpinalHDL/SpinalHDL/blob/b6368aaed142fccf51b1b0fea777eb3f911d40da/core/src/main/scala/spinal/core/internals/ComponentEmitterVerilog.scala#L1018

Best Regards.

Dolu1990, 2021-06-11:
Hi,

So, whould we either 
1) allow negative numbers in initBigInt (with the proper generation) ?
2) Or we should just catch it as a user error ?

selendym, 2021-06-11:
I'd suggest allowing it like the VHDL part already does. Consider, for example:
```scala
scala> BigInt( 0x80000000 )
val res0: scala.math.BigInt = -2147483648
```
and
```scala
scala> BigInt( 0x80000000L )
val res1: scala.math.BigInt = 2147483648
```
Most likely the latter was intended, so catching this as an error might be less than optimal. Some might also want to write `BigInt( -1 )` for all `1`s without thinking about the width, perhaps.

selendym, 2021-06-11:
It would seem that most if not all work has already been done with `initAssignmentBuilder` but the Verilog emitter seems to never actually use this value.

selendym, 2021-06-11:
On a second thought, perhaps it would be better to simply give an error for negative values. While `BigInt( -1 )` might be clear in intention, `BigInt( 0x80000000 )` should be considered an error, since it is not clear in this case whether or not to zero or sign-extend the value.

selendym, 2021-06-11:
Something like the following might work reasonably well when doing the conversion:
```scala
def bigInt2BitString( value: BigInt, bitWidth: Int, allowNegative: Boolean = false ): String = {
    assert( bitWidth > 0 && bitWidth >= value.abs.bitLength )  // NOTE: `value.abs.bitLength` may be `0`, so explicit `bitWidth > 0` is needed.
    assert( allowNegative || value >= 0 )

    val value_ =
        if ( allowNegative ) { (BigInt( 1 ) << bitWidth) + value }
        else { value }
    val bitString = value_.toString( 2 ).reverse.padTo( bitWidth, "0" ).take( bitWidth ).reverse.mkString

    return bitString
}
```
In essence, negative values result in an error by default but they can be allowed when the need arises.

Some examples:
```scala
scala> bigInt2BitString( BigInt( 0x80000001 ), 33 )
java.lang.AssertionError: assertion failed
  at scala.Predef$.assert(Predef.scala:264)
  at bigInt2BitString(<console>:3)
  ... 34 elided

scala> bigInt2BitString( BigInt( 0x80000001 ), 33, true )
val res128: String = 110000000000000000000000000000001

scala> bigInt2BitString( BigInt( 0x80000001L ), 33 )
val res129: String = 010000000000000000000000000000001

scala> bigInt2BitString( BigInt( -1 ), 33, true )
val res130: String = 111111111111111111111111111111111
```
Note specifically the differing MSB in the 2nd and 3rd example that might cause moments of WTF if not expected (by setting `allowNegative` to `true`).

In general, when `allowNegative` is `true`, the gloves are off and weird stuff like this might just happen:
```scala
scala> bigInt2BitString( BigInt( 0xe0000000 ), 32, true )
val res135: String = 11100000000000000000000000000000

scala> bigInt2BitString( BigInt( 0xe0000000 ), 30, true )
val res136: String = 100000000000000000000000000000

scala> bigInt2BitString( BigInt( 0xe0000000L ), 30, true )
java.lang.AssertionError: assertion failed
  at scala.Predef$.assert(Predef.scala:264)
  at bigInt2BitString(<console>:2)
  ... 34 elided
```
It might be expected that `0xe0000000` and `0xe0000000L` have the same bit widths but, alas, this is not true, so the last one fails correctly.

(Note that I haven't done extensive testing, so there might be bugs.)

@Dolu1990
Do you think that above would be reasonable? That is, error by default but allow when needed.

Dolu1990, 2021-06-14:
Hi @selendym,

I would say let's avoid as much of possible using string for it.

Instead, i just added a allowNegative argument to the initBigInt function and  a assert that everything is not negative if it isn't set.

That way there is a easy workaround and the generation backend only has to manage natural number.

is that good ?

closed 2021-06-14 by Dolu1990

selendym, 2021-06-14:
Sure, that works too. In that case, you might also want to revisit the memory emitters at some point and force this new restriction with asserts or similar. There should also be some room for refactoring/simplification - at least the Verilog emitter seemed to have some dead code there.

Dolu1990, 2021-06-14:
Right, it is quite possible XD

commit 3fcdff77 (2021-06-14): fix #442 (initBigInt with negative number)