SpinalHDL/SpinalHDL #445: Support In-line Verilog

volatile-static, 2021-06-15:
Is Spinal going to support black boxes with inline Verilog like Chisel?
```scala
import chisel3._
import chisel3.util.HasBlackBoxInline
class BlackBoxRealAdd extends BlackBox with HasBlackBoxInline {
  val io = IO(new Bundle {
    val in1 = Input(UInt(64.W))
    val in2 = Input(UInt(64.W))
    val out = Output(UInt(64.W))
  })
  setInline("BlackBoxRealAdd.v",
    """module BlackBoxRealAdd(
      |    input  [15:0] in1,
      |    input  [15:0] in2,
      |    output [15:0] out
      |);
      |always @* begin
      |  out <= $realtobits($bitstoreal(in1) + $bitstoreal(in2));
      |end
      |endmodule
    """.stripMargin)
}
```

Dolu1990, 2021-06-15:
Hi

It could be done, doesn't seem a big deal to implement. but more generaly, what's about inline verilog ? Allowing to insert a "blackboxExpression" into the AST, which will emit a user raw code into the netlist ?

I mean, what are the usages of black boxes with inline verilog ?

As far as i can see, emiting very specific chunk of verilog code, right ?

volatile-static, 2021-06-15:
Hi
I haven't given the question much thought. How do people use this in Chisel? Maybe someone will need it in the future 😄

volatile-static, 2021-06-15:
Imagine you're working on a project in pure SpinalHDL and all RTL files are in the gitignore. However, there is a little logic hard to be described in Spinal.
The point is that, can Verilog briefer than Spinal describing such a logic? 
For example, I have a switch matrix scanner circuit.
```verilog
always @(posedge clock, posedge reset)
	if (reset)
		key_down = 0;
	else begin: key_decode
		key_down = key_down;
		case (row_cnt)
			1: begin
				key_down[1] = col_sync[0];
				key_down[16:2] = col_sync[16:2];
			end
			2: begin
			  	key_down[37:17] = col_sync[20:0];
			end
			3: begin
			  	key_down[58:38] = col_sync[20:0];
			end
			4: begin
			  	key_down[70:59] = col_sync[11:0];
				key_down[71] = col_sync[13];
				key_down[74:72] = col_sync[19:17];
			end
			5: begin
			  	key_down[85:75] = col_sync[10:0];
				key_down[86] = col_sync[13];
				key_down[87] = col_sync[15];
				key_down[90:88] = col_sync[19:17];
			end
			0: begin
				key_down[93:91] = col_sync[2:0];
				key_down[94] = col_sync[5];
				key_down[96:95] = col_sync[10:9];
				key_down[101:97] = col_sync[17:13];
				key_down[103:102] = col_sync[20:19];
			end
			default: key_down = key_down;
		endcase
	end
```
It is synthesized like this:
[image]
each flip-flop has an enable port driven by the decoder with the order in `case`.
Yet I don't know how to describe such a circuit in SpinalHDL. 🙈

Dolu1990, 2021-06-16:
else begin: key_decode ?

Does does that do ?

What would be hard specificaly ?

volatile-static, 2021-06-16:
> Does does that do ?

It's no use in fact.
> What would be hard specificaly ?

I don't know exactly..

Dolu1990, 2021-06-16:
I'm not against the idea ^^
I just not sure if that should be integrated into a blackbox, or directly as a inlined verilog chunk in a module.

Dolu1990, 2021-06-16:
For the example you sent before, in Chisel you can't assign a sub part of a signals, but in SpinalHDL you can.

ex : xxx(2 downto 1) := 0

volatile-static, 2021-06-16:
> For the example you sent before, in Chisel you can't assign a sub part of a signals, but in SpinalHDL you can.
> 
> ex : xxx(2 downto 1) := 0

Yes, I know that. I mean, that will likely generate mux, instead of enable port of FF.

> I'm not against the idea ^^

Sorry, I misunderstood. I just don't know what to say.

> I just not sure if that should be integrated into a blackbox, or directly as a inlined verilog chunk in a module.

I think it better look like the inline assembly in C lang.

Dolu1990, 2021-06-16:
It should generate the same code than the verilog you showed, i guess.

volatile-static, 2021-06-16:
> It should generate the same code than the verilog you showed, i guess.
You're right. I got it. Thanks!

So, this issue will stay open until a linked pr is merged?

closed 2021-12-29 by volatile-static