SpinalHDL/SpinalHDL #448: Scala 2.13?

piegamesde, 2021-06-16:
I'm not really into the Scala ecosystem, but as far as I can tell Spinal needs to support specific Scala versions and it does not support 2.13. Is there any reason for this?

Dolu1990, 2021-06-18:
Hi,

So basicaly, scala 2.13 was breaking some minor parts of the API and is more like a scala version to use in order to upgrade to scala 3. To it was delayed.

Currently the plan is :
- Wait scala 3 to be "mature enough", it may actualy already be the case, i just checked a few issue i had by the past, they seems to be fixed
- Wait that we have enough time
- Upgrade the build.sbt to handle multiple version of scala in a single branch
- port dev (2.11) to 2.12 / 2.13 / 2.14
- then branch dev in a scala 3 version

@typingArtist  Miaouu

typingArtist, 2021-06-18:
Upgrading ```build.sbt``` is actually little work. What will be more work is to compare the 2.11 (dev) and 2.12 branches and unify these on the dev branch such that the cross compilation/building works.
In the past there was work done on SpinalHDL for 2.13 on a separate branch. Work came to a stall some two years ago. As this work is outdated, development should be picked up from the 2.12 branch, or from the dev branch once that is properly cross-building for 2.12.

piegamesde, 2021-06-18:
And I thought maintaining a project across multiple Java releases was a pain …

Dolu1990, 2021-06-18:
Not talking of the Scala compiler plugin  used in SpinalHDL (idslplugin) <3

piegamesde, 2021-07-14:
Seems to work with Spinal 1.6

closed 2021-07-14 by piegamesde