SpinalHDL/SpinalHDL #465: Axi4.incr causes Verilator error

JobNoorman, 2021-07-07:
`Axi4.incr` [contains](https://github.com/SpinalHDL/SpinalHDL/blob/eae02ef049ed0c0c3bd8ece4d06c0b0f12f02d1c/lib/src/main/scala/spinal/lib/bus/amba4/axi/Axi4.scala#L259-L264) the following mux:

```scala
val wrapCase = validSize.resize(wrapCaseWidth) + len.mux(
  M"----1---" -> U"11",
  M"-----1--" -> U"10",
  M"------1-" -> U"01",
  default -> U"00"
)
```
Since #437 this is compiled to the following Verilog:

```verilog
always @(*) begin
  casez(unburstify_buffer_len)
    8'b????1??? : begin
      _zz_Axi4Incr_wrapCase = 2'b11;
    end
    8'b?????1?? : begin
      _zz_Axi4Incr_wrapCase = 2'b10;
    end
    8'b??????1? : begin
      _zz_Axi4Incr_wrapCase = 2'b01;
    end
    default : begin
      _zz_Axi4Incr_wrapCase = 2'b00;
    end
  endcase
end
```

Which causes an error in Verilator:
```
%Warning-CASEOVERLAP: /.../tmp/job_1/Test.v:253: Case values overlap (example pattern 0xc)
      8'b?????1?? : begin
      ^~~~~~~~~~~
                      ... Use "/* verilator lint_off CASEOVERLAP */" and lint_on around source to disable this message.
%Error: Exiting due to 1 warning(s)
```

Note that this makes `Axi4SharedOnChipRam` unusable in 1.5.0.

Here is a minimal working example that generates this error:
```scala
object Test {
  def main(args: Array[String]) {
    SimConfig.withWave.compile(Axi4SharedOnChipRam(
      byteCount = 4 KiB,
      dataWidth = 32,
      idWidth = 4
    ))
  }
}
```

Dolu1990, 2021-07-08:
I pushed a fix to disable the emition of casez on MaskedLiteral.

More generaly, i'm not sure if verilator consider that casez should not be used, or if i missed something.

If somebody knowes, let us know :)

JobNoorman, 2021-07-08:
I don't think the _use_ of `casez` is a problem for Verilator, it's the fact that cases overlap. This is what the manpage has to say about it:

> CASEOVERLAP
>
> Warns that inside a case statement you have case values which are detected to be overlapping.  This is bad style, as moving the order of case values will cause different behavior.  Generally the values can be respecified to not overlap.
> 
> Ignoring this warning will only suppress the lint check, it will simulate correctly.

The mux used in `Axi4.incr` has overlapping cases at the SpinalHDL level, and it relies on the fact that `mux` will select the first match (although this doesn't seem to be specified anywhere in the docs). Since this is also the behavior of `casez` in Verilog, it might also be an option to simply disable this warning (using `verilator lint_off CASEOVERLAP`).

nbstrike, 2021-07-09:
Isn't it the case that the mux here is under-specified? ie shouldn't it be something more like:
```scala
val wrapCase = validSize.resize(wrapCaseWidth) + len.mux(
  M"----1---" -> U"11",
  M"----01--" -> U"10",
  M"----001-" -> U"01",
  default -> U"00"
)
```
Without using casez then I think this would have been implied by the chain of if/else-if checks.

nbstrike, 2021-07-09:
Come to think of it, perhaps this would be most neatly coded as:
```scala
   val wrapCase = validSize.resize + True.mux(len(3) -> U"11", len(2) -> U"10", len(1) -> U"01", default -> U"00")
```

The only other location I can see in the code base that would seem to be susceptible to this problem is https://github.com/SpinalHDL/SpinalHDL/blob/ab9817c528061e2ed65629367a0170dd5ce5d5e5/tester/src/main/scala/spinal/tester/PlayDev.scala#L119-L133

Dolu1990, 2021-07-09:
Thanks @JobNoorman @nbstrike 

I bringed back the masked literal in the verilog switch statement and just applied the nbstrike mux fix :)

https://github.com/SpinalHDL/SpinalHDL/commit/31a27596e77ed38c685d42f797bad40c72e62202

Dolu1990, 2021-07-09:
That PlayDev.scala is just a sandbox to play arround, no worries there

closed 2021-07-09 by Dolu1990

JobNoorman, 2021-07-10:
Thanks @Dolu1990 for the quick fix! I have two questions:
1. I think it might be a good idea to generate an error at the SpinalHDL level for overlapping cases as Verilator errors can be very confusing to interpret. Would this be possible?
2. Would it be possible to make a patch release that includes this fix? I'd really like to start using 1.5.0 but this issue is blocking for me.

Dolu1990, 2021-07-10:
@JobNoorman I'm currently about to release 1.6.0

and yes, more in depth checks would be possible to add. Duplication on raw literals are already checked, but not masked literals

Dolu1990, 2021-07-10:
1.6.0 released, it will break one thing : To define a Bool you need now to write Bool()