SpinalHDL/SpinalHDL #467: A little bit of documentation or explanation!

akshar001, 2021-07-09:
We were trying to run the workshop example StreamUnit yesterday.
We were figuring out what it does?

```
  val memReadStream = mem.streamReadSync(io.cmdA)
  io.rsp << StreamJoin.arg(memReadStream,io.cmdB).translateWith(memReadStream.payload ^ io.cmdB.payload)
```

Three questions,
Why used xor?
Why used translatewidth?
What do the stream really does?

As per our understanding, it's used for moving one data from one modules to another modules.
So it's serializing and sending or it's parallel sending?

I know i sound dumb, but we want to understand full scale spinalhdl , We know the python stuff we know the verilog stuff but not Scala and your concept. Can you anyway help us to understand each of them?

Dolu1990, 2021-07-09:
Hi,

No worries about the question :)

So, 
> Why used xor

memReadStream.payload ^ io.cmdB.payload  is just to have some simple computation in the exercise, there is no practical reasons in it, excepted it is a simple operation to do.

> translatewidth

val valueB = xxx
streamA.translateWith(valueB)

See https://github.com/SpinalHDL/SpinalHDL/blob/dev/lib/src/main/scala/spinal/lib/Stream.scala#L295

Basicaly, it return a stream combinatorialy connected from streamA, but with it's payload replaced with valueB.
It is a simple way to mutate the payload of a stream.

> What do the stream really does?

A Stream is a simple arbitration capabile of carrying a payload :
https://spinalhdl.github.io/SpinalDoc-RTD/SpinalHDL/Libraries/stream.html?highlight=stream

The presence of a payload is signaled from the master with valid := True, while the slave can accept it by signaling ready := True

akshar001, 2021-07-10:
And we spent 4 hours that what should be the mean of it, 
We ran scala simulation tester, have some ideas.

closed 2021-07-10 by akshar001

akshar001, 2021-07-10:
@Dolu1990 we want to have an basic explanation for all of the things. Can you a create any discord channel or can we have a session for understanding a code. We can pay for hours ( if it is required )  but it would be helpful for understand a basic concepts for us which is taking a time.

Dolu1990, 2021-07-10:
@akshar001 We can organize that. Can you contact me with charles.papon.90@gmail.com ?
One thing, from the 12 july to the 19 july i will be in a holiday travel, so hardly joinable.