SpinalHDL/SpinalHDL #468: Verilator report error due to `case values overlap`

wswslzp, 2021-07-11:
Hi, 

Recently SpinalHDL has updated from 1.4.3 to 1.5.0 and we see that it has improved the Verilog generation. However, when I rerun the test case which works well in the previous version, now Verilator reports some warnings and errors due to the `case values overlap`. 

The version of Verilator is 4.101 devel rev v4.100-38-g0a9ae15.

Is there a way to fix or work around the problem?

```
[warn] multiple main classes detected: run 'show discoveredMainClasses' to see the list
[warn] multiple main classes detected: run 'show discoveredMainClasses' to see the list
[info] running (fork) FpgaTest.NlosCoreTest
[info] [Runtime] SpinalHDL v1.5.0    git head : ef932e10212ecc51970e3f01c43e97d4eb4bb2df
[info] [Runtime] JVM max memory : 14564.0MiB
[info] [Runtime] Current date : 2021.07.11 10:19:52
[info] [Progress] at 0.000 : Elaborate components
[info] [Progress] at 2.662 : Checks and transforms
[info] [Progress] at 20.111 : Generate Verilog
[info] [Warning] 4266 signals were pruned. You can call printPruned on the backend report to get more informations.
[info] [Done] at 25.058
[info] [Progress] Simulation workspace in /home/Data1/Zhengpeng/work/nlos/nlos_system/tb/NlosCore
[info] [Progress] Verilator compilation started
[info] %Warning-CASEOVERLAP: /home/Data1/Zhengpeng/work/nlos/nlos_system/tmp/job_1/NlosCore.v:150669:7: Case values overlap (example pattern 0xc)
[info]                                                                                                : ... In instance NlosCore.rgca.impulse_load_unit
[info] 150669 |       8'b?????1?? : begin
[info]        |       ^~~~~~~~~~~
[info]                       ... Use "/* verilator lint_off CASEOVERLAP */" and lint_on around source to disable this message.
[info] %Warning-CASEOVERLAP: /home/Data1/Zhengpeng/work/nlos/nlos_system/tmp/job_1/NlosCore.v:152026:7: Case values overlap (example pattern 0xc)
[info]                                                                                                : ... In instance NlosCore.rgca.wave_load_unit
[info] 152026 |       8'b?????1?? : begin
[info]        |       ^~~~~~~~~~~
[info] %Warning-CASEOVERLAP: /home/Data1/Zhengpeng/work/nlos/nlos_system/tmp/job_1/NlosCore.v:152428:7: Case values overlap (example pattern 0xc)
[info]                                                                                                : ... In instance NlosCore.rgca.distance_load_unit
[info] 152428 |       8'b?????1?? : begin
[info]        |       ^~~~~~~~~~~
[info] %Warning-CASEOVERLAP: /home/Data1/Zhengpeng/work/nlos/nlos_system/tmp/job_1/NlosCore.v:152683:7: Case values overlap (example pattern 0xc)
[info]                                                                                                : ... In instance NlosCore.rgca.timeshift_load_unit
[info] 152683 |       8'b?????1?? : begin
[info]        |       ^~~~~~~~~~~
[info] %Error: Exiting due to 4 warning(s)
[error] Exception in thread "main" java.lang.AssertionError: assertion failed: Verilator invocation failed
[error]         at scala.Predef$.assert(Predef.scala:223)
[error]         at spinal.sim.VerilatorBackend.compileVerilator(VerilatorBackend.scala:487)
[error]         at spinal.sim.VerilatorBackend.<init>(VerilatorBackend.scala:553)
[error]         at spinal.core.sim.SpinalVerilatorBackend$.apply(SimBootstraps.scala:135)
[error]         at spinal.core.sim.SpinalSimConfig.compile(SimBootstraps.scala:637)
[error]         at spinal.core.sim.SpinalSimConfig.compile(SimBootstraps.scala:600)
```

Thanks.

wswslzp, 2021-07-11:
The generated Verilog code segment is 

```
  always @(*) begin
    casez(aw_area_awlen_r)
      8'b????1??? : begin
        _zz_Axi4Incr_wrapCase = 2'b11;
      end
      8'b?????1?? : begin
        _zz_Axi4Incr_wrapCase = 2'b10;
      end
      8'b??????1? : begin
        _zz_Axi4Incr_wrapCase = 2'b01;
      end
      default : begin
        _zz_Axi4Incr_wrapCase = 2'b00;
      end
    endcase
  end
```

These codes occur in four different modules. It seems that there's no overlapping in the generated code.

wswslzp, 2021-07-11:
The above Verilog code segment is generated using `Axi4.incr()` function.

closed 2021-07-11 by wswslzp

Dolu1990, 2021-07-11:
so 1.6.0 fixed it, but be aware you will have to patch each Bool into Bool()

wswslzp, 2021-07-11:
Thanks. Just seeing the update :)