SpinalHDL/SpinalHDL #475: Feature request: Reuse compiled binaries

jonnykl, 2021-07-15:
I'm working on a large project where the verilator compilation needs up to 30 minutes. Often I only make changes to the test - leaving the main source code unchanged. SpinalHDL always recompiles everything though it isn't needed (I guess).

It would be useful if SpinalHDL would automatically reuse the compiled binaries. Alternatively it would be helpful I there were a way to target a directory containing the compiled binaries which are then used. Then one could run the test normally in the first place and then set the option targeting the directory containing the binaries.

Dolu1990, 2021-07-20:
Hi, 

> SpinalHDL always recompiles everything though it isn't needed (I guess).

Yes that's right

Seems like verilator automaticaly use ccache if you have it already installed on your PC, this can already save some time. But still, the Verilator translation from verilog to C++ code can take quite a while and seems to be largely monothreaded. 

To improve this we would need to cache the whole verilator process. That would be easily fesable i guess, and quite usefull, i agree.

This is where verilator is asked to generate the C++ model : 
https://github.com/SpinalHDL/SpinalHDL/blob/dev/sim/src/main/scala/spinal/sim/VerilatorBackend.scala#L486

and where it compile the C++ into a .so : 
https://github.com/SpinalHDL/SpinalHDL/blob/dev/sim/src/main/scala/spinal/sim/VerilatorBackend.scala#L491

I'm currently in a total rush for a few deadlines, so i can't handle that in a short time.

jonnykl, 2021-07-20:
> Seems like verilator automaticaly use ccache if you have it already installed on your PC, this can already save some time. But still, the Verilator translation from verilog to C++ code can take quite a while and seems to be largely monothreaded.

Just tried this, it works but the verilator translation still needs a lot of time.
 
> To improve this we would need to cache the whole verilator process. That would be easily fesable i guess, and quite usefull, i agree.

I implemented a simple cache for the verilator binaries, its not perfect but it works. See #477

PR #477 (2021-08-05): add cache for verilator binaries
Creates a cache for the binaries generated by verilator. The hash of the verilator options and the source file contents will be saved and compared before verilator gets invoked. If the hashes equal, the binaries are copied from the cache and verilator compilation will be skipped.

This works for me for some components. SpinalHDL does not always generate the same RTL given the same scala source code. In this case the cached is not used (because the hash always changes). A possible fix (besides fixing SpinalHDL to generate the same RTL given the same scala source [... cut]
closed 2021-08-05 by Dolu1990