SpinalHDL/SpinalHDL #476: Reg_bundle_getzero_problem

dreamflyings, 2021-07-20:
When I was writing test code, I found an interesting question.
// todo vtype_t is a bundle
    val rst = ClockDomain.current.reset.pull()
    val tmp = (vtype_t().getZero).setAsReg().allowOverride // todo It's work, tmp can be reset zero
// val tmp = Reg(vtype_t().getZero).allowOverride // todo This way doesn't work
// val tmp = Reg(vtype_t()).getZero.allowOverride // todo Can't work well
    when(!rst){
      tmp.assignFromBits(B"xffff")
    }/*.otherwise{
      tmp.assignFromBits(B"xA0A0")
    }*/

Dolu1990, 2021-07-21:
Hi,

I'm not sure to understand, what is the expected behaviour and the current behaviour ?

dreamflyings, 2021-07-23:
@Dolu1990 
Hello, Dolu1990 ,thank you very much for your reply, thank you for your outstanding contribution to spinalHDL. Thank you also for writing such an incredible software framework like Vex.
This question is just to start with, whatever the setting is acceptable, it is best to achieve logical consistency.
I am currently working on converting systemverilog to spinal.
When rewriting sv, I encountered a lot of code problems for setting the initial value and setting the reset value. Rewriting with spinal is very cumbersome, and it is also difficult to write scripts.
It is recommended to think more about the compatibility of spinal with chisel and systemverilog in the future (the code can be used after a simple script processing),
For example, Vec(Int, =>T) support, .W support, **support, asTypeOf(in ChiselTypeCast), etc. In addition, it would be better to fully support Enum used in sv, next(), prev(), inside, etc. These can maximally assist programmers in using/copying existing codes at low cost.
Currently Spinal supports: Mem, Bool, UFix, SFix, SpinalEnum, SInt, UInt type data initialization. There is no simplified way for Bundle and Vec to set the initial value and set the reset value.
E.g:
Set the initial value of struct in systemverilog:
`
always_comb begin
    acc_resp_o ='{
      id: id,
      complete: isZero | complete_i,
      pending: pending_i,
      exValid: |ex_valid_i,
      default: '0
    };
end
`
Set the reset value of struct in systemverilog:
`
  always_ff @(posedge clk_i or negedge rst_ni) begin
    if (!rst_ni) begin
      acc_resp_o ='{
      pending: {8{1'b1}},
      default: '0
    };
    end else begin
      ???
    end
  end
`
Set initial value in spinal
`
    acc_resp_o.assignFromBits(B(0,acc_resp_o.getBitsWidth bits))
//acc_resp_o.assignFromBits(B(default -> 0)) //todo it's error
    acc_resp_o.id := id
    acc_resp_o.complete := isZero | complete_i
    acc_resp_o.pending := pending_i
    acc_resp_o.exValid := ex_valid_i.orR
`
Set the reset value in spinal
`
val accTmp = acc_respBundle()
accTmp.assignFromBits(B(0,acc_resp_o.getBitsWidth bits))
accTmp.pending := (default -> True)
acc_resp_o.init(accTmp)
OR:
acc_resp_o.assignFromBits(B(0,acc_resp_o.getBitsWidth bits))
//acc_resp_o.assignFromBits(B(default -> 0)) //todo it's error
acc_resp_o.assignSomeByName(new Bundle{
val id = tmpU.asBits.asUInt // todo genVerilog => {tmpU_7,{tmpU_6,{tmpU_5,{tmpU_4,{tmpU_3,{tmpU_2,{tmpU_1,tmpU_0}}}}}}}
val complete = isZero | complete_i
val pending = pending_i
val exValid = ex_valid_i.orR
})
`
Requirements: (or other formats or function names, mainly to facilitate script processing)
`
acc_resp_o.setValue{
      id -> id,
      complete -> isZero | complete_i,
      pending -> pending_i,
      exValid -> ex_valid_i.orR,
      default -> 0
}
acc_resp_o.init{
      pending -> (default -> true),
      default -> 0
}
`
General
`
SomeBundle.setValue{
someField -> someValue,
subBundle -> subBundle.setValue(...)
     default -> elseValue
}
SomeBundle.init{
someField -> someValue,
subBundle -> subBundle.setValue(...)
     default -> elseValue
}
`
If time permits, I would like to rewrite Xiangshang (with two thousand stars in just one month) or the upgraded version of the project at the end of the year into spinal, so please take into consideration that there are engineers who have ideas like me (copy chisel 2 spinal) Appeal. (As some colleges and universities in China start to use Chisel, the gap between our spinal and Chisel has begun to widen. It may be a wise move to be more compatible with SV and Chisel)
Thank you again.

Dolu1990, 2021-07-27:
Hi,

> Set the initial value of struct in systemverilog:
always_comb begin acc_resp_o ='{ id: id, complete: isZero | complete_i, pending: pending_i, exValid: |ex_valid_i, default: '0 }; end

So currently, there is no auomated way to have a function to provide similar initialisation behaviour, you have two ways : 
- setting things manualy one by one
- Manualy define a function which does the same

We could imagin a scala macro or a scala compiler plugin to add that functionality

> val accTmp = acc_respBundle() accTmp.assignFromBits(B(0,acc_resp_o.getBitsWidth bits))

can also be 

val accTmp = acc_respBundle().getZero //getZero return a new Bundle with all members assigned to zero

> Set the reset value in spinal

On thing, is you can call the init function on Bundle members.
You can also use the init(myBundle.getZero)

> SomeBundle.init{ someField -> someValue, subBundle -> subBundle.setValue(...) default -> elseValue }
> SomeBundle.setValue{ someField -> someValue, subBundle -> subBundle.setValue(...) default -> elseValue }

my guess is that some features like this could be implemented via macro / compiler plugin.

That would avoid : 

SomeBundle.someField init(someValue)
SomeBundle.someField2 init(someValue2)
SomeBundle.someField3 init(someValue3)

> the gap between our spinal and Chisel has begun to widen

which gap ?

numero-744, 2022-10-13:
I think the widening gap it is that Chisel and Spinal syntax differ more and more (but is SpinalHDL supposed to just be a Chisel fork?).

I think this issue can be closed, feel free to reopen it or open other issues if needed.

(I am curious about the SV-to-SpinalHDL converter, especially about clock domains and the maintainability of the generated code, vs using just Blackboxes).

Dolu1990, 2022-10-13:
> (but is SpinalHDL supposed to just be a Chisel fork?).

It isn't ^^
especialy since Chisel 3.