SpinalHDL/SpinalHDL #482: todo : remove not used signal on port

jijingg, 2021-07-27:
@Dolu1990 
can we remove not used signal on Component port or  even removed not used Component. because those code will not cause  `NO DRIVER ON  ....  [input]/[output]` Error !  but cause Lint error in tradition IC flow .

i try'd removePruned  option on SpinalConfig , but nothing happen .

```scala 
  class Sub extends Component{
    val io = new Bundle{
      val x = in Bits(32 bit)
      val y = out Bits(32 bit)
    }
  }
  class Top extends Component{
    val io = new Bundle{
      val din = in Bits(32 bit)
      val dout = out Bits(32 bit)
    }

    val notused1, notused2 = Bits(32 bit)
    val dut = new Sub()

    dut.io.x  := notused1. 
    notused2 := dut.io.y
    io.dout := io.din
  }
```

it will generate the code unobstructed and without warning
the port `x`,`y` should be removed , if all port be removed ,the component Sub need be removed too!.
```verilog
module Top (
  input      [31:0]   io_din,
  output     [31:0]   io_dout
);
  wire       [31:0]   dut_io_y;
  wire       [31:0]   notused1;
  wire       [31:0]   notused2;

  Sub dut (
    .io_x    (notused1[31:0]  ), //i
    .io_y    (dut_io_y[31:0]  )  //o
  );
  assign notused2 = dut_io_y;
  assign io_dout = io_din;

endmodule

module Sub (
  input      [31:0]   io_x,
  output     [31:0]   io_y
);

endmodule 
```

Dolu1990, 2021-07-28:
Hi,

I'm ok with it being implemented as a option, but turned off by default. The reason is that durring the multiple iterations that the user  design his thing, it is very easy to forget connecting something. The issue if SpinalHDL remove those unused named signal is that things will magicaly disapear, at the surprise of the user.

So, ok for it as a option, but disabled by default :)

and while looking at it, i found out that there is a option to do that already : 

  SpinalConfig(removePruned = true).generateVerilog(new Top)
  
  It should do what you want i guess. let's me know.

jijingg, 2021-07-28:
> So, ok for it as a option, but disabled by default :)
and while looking at it, i found out that there is a option to do that already :
SpinalConfig(removePruned = true).generateVerilog(new Top)

yes , wonderful,  that's work 👍 , i make mistake before use this option .

jijingg, 2021-07-28:
@Dolu1990 Another thing, I want to confirm
Is there any  `keep flag` to keep the signals I want to moniter and dont removed when I turn on the `removePruned = true` option?

jijingg, 2021-07-29:
The main scenarios for users  reference when option removePruned on
 
[image]

closed 2021-07-29 by jijingg