SpinalHDL/SpinalHDL #486: Repeating "_1" suffixes of the name of verilator working directory

wswslzp, 2021-08-02:
When using the `spinal sim` to do the regression tests on the same dut with verilator backend,  spinal will rename the simulation workspace by adding the suffix `_1` after the previous one, in order to avoid the naming conflict. However, the name will soon be lengthy if there are many test cases to run, shown as below
[image]
Maybe replacing the repeating `_1` with `_<incremental_number>`, e.g. `_1_1` with `_2`, is more reasonable and expected.

Dolu1990, 2021-08-03:
Ahh, i was able to reproduce the case when reusing the same SimConfig multiple times. I pushed a fix. Can you give a try and let me know if things are also fixed for you ?

wswslzp, 2021-08-03:
Thanks very much for your works, the issue is fixed!

closed 2021-08-03 by wswslzp