SpinalHDL/SpinalHDL #488: Vec_Bundle_for Assignment problem

dreamflyings, 2021-08-10:
Hello ! 
When I was rewriting the SV code, I found some unexplainable phenomena. 
The correct and runnable code is on the next line of todo.
 And the variable result_d in the second phenomenon does not need to use //.noCombLoopCheck.
 Are these two phenomena a mechanism or a BUG. 
 It feels really difficult to write Spinal code as you like!
 ```scala
 case class TellMeWhy() extends Component{
  case class payload_t() extends Bundle {
    val id = UInt(4 bits)
    val addr = UInt(32 bits)
    val wdata = UInt(64 bits)
    val be = UInt(8 bits)
  }
  val cond = in(Bool())
  val read_pointer_n = UInt(3 bits)
  // todo The correct code is as follows
  //  val read_pointer_n = UInt(3 bits).noCombLoopCheck
  val read_pointer_q = Reg(UInt(3 bits)).init(0)
  read_pointer_n := read_pointer_q
  when(read_pointer_n === 7) {
    read_pointer_n := 0
  } .otherwise {
    read_pointer_n := read_pointer_q+1
  }
  when(cond) {
    read_pointer_q := 0
  } .otherwise {
    read_pointer_q := read_pointer_n
  }
  case class bitsTests_B() extends Bundle{
    val a = Bool()
    val b = Bits(8 bits)
  }
  val bitsTests = out(Vec(4, bitsTests_B()) ).setAsReg()
  bitsTests.map(_.init(bitsTests_B().getZero))
  bitsTests.allowUnsetRegToAvoidLatch
  val tmp_read_pointer = Cat(read_pointer_n,read_pointer_q,True,False)
  val vecSel = in(UInt(2 bits)).dontSimplifyIt().keep()
  for( i<- 0 until 4){
    switch(vecSel){
      is(0) { bitsTests(0).b(i) := tmp_read_pointer(i) }
      is(1) { bitsTests(1).b(i) := tmp_read_pointer(i) }
      is(2) { bitsTests(2).b(i) := tmp_read_pointer(i) }
      is(3) { bitsTests(3).b(i) := tmp_read_pointer(i) }
    }
    when(cond){
      bitsTests(3).b(4)  := tmp_read_pointer(4)
      bitsTests(3).b(5)  := tmp_read_pointer(5)
      bitsTests(3).b(6)  := tmp_read_pointer(6)
      bitsTests(3).b(7)  := tmp_read_pointer(7)
    }
  }
  val ResultDepth = 4
  val result_d = Wire(Vec(ResultDepth, Vec(8, payload_t())))//.noCombLoopCheck.allowOverride
  val result_q = Reg(Vec(ResultDepth, Vec(8, payload_t())))
  //  result_q.map(_.map(_.init(payload_t().getZero)))
  result_q := result_d
  result_d := result_q

  val beTest_cnt_i = in UInt(3 bits)

  for(axi_byte <- 0 until 8){
    //      result_d(0)(0).wdata(8*axi_byte,8 bits) := U"8'h47"
    //      (result_d(0)(0).be)(axi_byte) := cond | acc_in.insn.instr.msb
    result_d(read_pointer_q.resized)(beTest_cnt_i).wdata(8*axi_byte,8 bits) := U"64'h47"
    result_d(read_pointer_q.resized)(beTest_cnt_i).be(axi_byte) := (cond | vecSel.msb)
    // todo The correct code is as follows
    //  result_d(read_pointer_q.resized)(beTest_cnt_i).be(axi_byte, 1 bits) := (cond | vecSel.msb).asUInt.resized
  }
  val result = out(result_q)
}

object Test1 extends App{
  SpinalSystemVerilog(SpinalConfig(
    targetDirectory = "build/tests"
  )){
    val tmp = TellMeWhy()
    tmp
  }
}
 ```

dreamflyings, 2021-08-10:
Another solution to the second phenomenon is to set be to Vec.

Dolu1990, 2021-08-10:
Hi,

The code sample, copy pasted, has some chisel stuff in it (Wire), can you send a fully SpinalHDL compilable one ?

Then about the question,  seems your issue is about, why noCombLoopCheck and allowOverride are required sometimes ?

dreamflyings, 2021-08-10:
> Hi,
> 
> The code sample, copy pasted, has some chisel stuff in it (Wire), can you send a fully SpinalHDL compilable one ?
> 
> Then about the question, seems your issue is about, why noCombLoopCheck and allowOverride are required sometimes ?

Thank you for your reply. This code works in my environment, and deleting Wire() should work. Part of the above code is indeed copied from the file after script processing, mainly for faster reproduction and problem solving. Basic understanding of how to use noCombLoopCheck and allowOverride, but in the process of using complex projects, there will often be some strange phenomena, some will be difficult to locate. For some sv open source code translation results, problems such as COMBINATORIAL LOOP, LATCH DETECTED, etc. often appear, but SV can run through the program, and these problems are difficult to locate. For example, sometimes, use the following code
```scala
val axi_ar_o = out(axi_ar_t().getZero())
// val axi_ar_o = out(axi_ar_t())
// axi_ar_o.getZero()
```
Instead of being annotated code is a good idea (reason unknown).
There are still many problems like this, which may be caused by the SV code style, but I also hope that spinal has better robust performance and is more compatible with SV code style.

Dolu1990, 2021-08-11:
Hi, 

So, there is so far 3 issues : 

1) There was some unimplemented stuff in SpinalHDL which made impossible to assign specific parts of a bits from a Vec using UInt as index for the vec : 

result_d(read_pointer_q.resized)(beTest_cnt_i).be(axi_byte) := (cond | vecSel.msb)

I just pushed a fix in dev  via e08d92e.

2) minor width missmatch, result_d(read_pointer_q.resized)(beTest_cnt_i).wdata(8*axi_byte,8 bits) := U"8'h47"       
 U"8'h47" instead of  U"64'h47"

3) "Paradigme" issue (COMBINATORIAL LOOP) : 

```scala

  val read_pointer_n = UInt(3 bits)
  ...
  when(read_pointer_n === 7) {
    read_pointer_n := 0
  }
``` 

This is a combinatorial loop. In SpinalHDL, combinatorial signal assignement behave like verilog "<=" assignement, (Nonblocking Assignment) and you have read_pointer_n depending on itself. If you want blocking assignement behaviour you need : 

```scala
  var read_pointer_n = UInt(3 bits)
  
  ..
  
    when(read_pointer_n === 7) {
    read_pointer_n \= U(0, 3 bits)
  } .otherwise {
    read_pointer_n \= read_pointer_q+1
  }
  
```

But i would say, this sementic isn't very much to be used in SpinalHDL

Or you need to define separated signals instead of reusing read_pointer_n.

> more compatible with SV code style

So, first, SpinalHDL in itself do not aim for SV code style, but I understand the need for compatibility. If there things which "suck" do not hesite to open an issue ^^

Dolu1990, 2021-08-11:
hmmmm for read_pointer_n

I guess the "proper" code style fix would be : 

```scala
  val read_pointer_n = UInt(3 bits)
  // todo The correct code is as follows
//    val read_pointer_n = UInt(3 bits).noCombLoopCheck
  val read_pointer_q = Reg(UInt(3 bits)).init(0)
  when(read_pointer_q === 7) { //here swap _n for _q
    read_pointer_n := U(0, 3 bits)
  } .otherwise {
    read_pointer_n := read_pointer_q+1
  }
  ```

dreamflyings, 2021-08-12:
@Dolu1990 
Thank you very much!
Not only replied to the points of concern in detail, but also quickly fixed the BUG.
I understand the usage of := and \=. However, \= can not act on the interface signal and its related signals in the impression, and using var and \= will modify a lot of codes, so the simple way currently used is to add the noCombLoopCheck method.
There are some small problems, I will take the time to push the ISSUE, thank You for the great efforts made by the cause spinal again.

numero-744, 2022-10-13:
I think it can be closed as completed, and other issues can be open for other stuff.

closed 2022-10-13 by Dolu1990