SpinalHDL/SpinalHDL #489: Component.rework breaks StateMachine

JobNoorman, 2021-08-11:
The following code fails with an obscure error message:

```scala
class Test extends Component {
  val fsm = new StateMachine {
    Test.this.rework()
    val entry = StateEntryPoint()
  }
}
```

The error message is:

```
toplevel/fsm as no entry point set. val yourState : State = new State with EntryPoint{...}   should solve the situation at
[...]
```

Which is very confusing since there clearly is an entry point.

Note that this only happens when `rework` is called before creating the entry state.

I think the problem is that `rework` calls `prePop` which causes the `StateMachine` to be built before its states are fully constructed.

Dolu1990, 2021-08-11:
Hi,

I guess you just hit one actual limitation of the rework function, it will trigger all addPrePopTask, including the one of the state machine.

I guess this prePpTask should be more procisely handeled by SpinalHDL
I will try to implement a fix.

closed 2021-08-11 by Dolu1990

JobNoorman, 2021-08-11:
Thanks a lot for the quick fix!

commit dbed0611 (2021-08-11): Fix #489 avoid Component.rework to execute afterElaboration tasks