SpinalHDL/SpinalHDL #495: Axi4SharedToApb3Bridge connection error

mjc0608, 2021-08-23:
Hi,

I am new to SpinalHDL and do not know whether this is a bug or a feature. I tried to instantiate an Axi4SharedToApb3Bridge and output the Verilog code with the following SpinalHDL code:

```scala
class Top(aw: Int,dw: Int,iw: Int) extends Component{
  val io = new Bundle {
    val axi = slave(Axi4Shared(Axi4Config(aw,dw,iw)))
    val apb = master(Apb3(Apb3Config(aw,dw,iw)))
  }
  val bridge = Axi4SharedToApb3Bridge(aw,dw,iw)
  io.axi <> bridge.io.axi
  io.apb <> bridge.io.apb
}
```

SpinalHDL gives me the following error:

```
[error] Exception in thread "main" spinal.core.SpinalExit: 
[error]  Can't zip [toplevel/io_axi_arw_payload : Axi4Arw] with [toplevel/bridge/io_axi_arw_payload : Axi4Arw]  because they don't have the same number of elements.
[error] First one has :
[error] (addr,(toplevel/io_axi_arw_payload_addr : in UInt[20 bits]))
[error] (id,(toplevel/io_axi_arw_payload_id : in UInt[4 bits]))
[error] (region,(toplevel/io_axi_arw_payload_region : in Bits[4 bits]))
[error] (len,(toplevel/io_axi_arw_payload_len : in UInt[8 bits]))
[error] (size,(toplevel/io_axi_arw_payload_size : in UInt[3 bits]))
[error] (burst,(toplevel/io_axi_arw_payload_burst : in Bits[2 bits]))
[error] (lock,(toplevel/io_axi_arw_payload_lock : in Bits[1 bits]))
[error] (cache,(toplevel/io_axi_arw_payload_cache : in Bits[4 bits]))
[error] (qos,(toplevel/io_axi_arw_payload_qos : in Bits[4 bits]))
[error] (prot,(toplevel/io_axi_arw_payload_prot : in Bits[3 bits]))
[error] (write,(toplevel/io_axi_arw_payload_write : in Bool))
[error] Seconde one has :
[error] (addr,(toplevel/bridge/io_axi_arw_payload_addr : in UInt[20 bits]))
[error] (id,(toplevel/bridge/io_axi_arw_payload_id : in UInt[4 bits]))
[error] (len,(toplevel/bridge/io_axi_arw_payload_len : in UInt[8 bits]))
[error] (size,(toplevel/bridge/io_axi_arw_payload_size : in UInt[3 bits]))
[error] (burst,(toplevel/bridge/io_axi_arw_payload_burst : in Bits[2 bits]))
[error] (write,(toplevel/bridge/io_axi_arw_payload_write : in Bool))
[error] ********************************************************************************
[error] ********************************************************************************
[error] Design's errors are listed above.
```

It seems like the signal generated by the bridge is less than the signal generated by the IO bundle.

Thanks,
Jiacheng

mjc0608, 2021-08-23:
It seems like these signals can be turned off in the config. So it's a feature.

closed 2021-08-23 by mjc0608

Dolu1990, 2021-08-24:
My guess is that instead of using <> to connect the axi, you can use 
  io.axi >> bridge.io.axi
  
  the >> operator is implemented in the axi class to take care of useless signals and do small adjustement to make things compatible

mjc0608, 2021-08-24:
Nice. Thanks!