SpinalHDL/SpinalHDL #496: Make state machine without entry/exit point (remove BOOT state)

piegamesde, 2021-08-24:
At the moment, every FSM synthesizes a `BOOT` state, that transitions into the entry point state on `start` and that will be entered on `finish`.

This totally makes sense for state machines that can be "activated" and "terminate", including but not limited to inner state machines (#330). However, for the use case of an infinitely running state machine that is never to exit, this is not necessary. It is wasteful (that extra state with transitions causes extra logic, that can not 100% be optimized away) and may even be harmful (as far as I can tell, it delays the initial state by one cycle).

Thus, I'd like to have the option to go without. This would of course mean that the machine has no `start` and `finish`, but it also has the subtle difference that the `onEntry` of the initial state won't be called on `reset`.

An alternative would be to make the BOOT state available through API, so that I can make my entry state the BOOT state.

Dolu1990, 2021-08-24:
I agree it would be a good thing to have the choice.

One thing to not forget, is that the BOOT state also provide a clean onEntry to the entry state. (one cycle before the entry state is active)

so, not having that boot state would kind of remove the onEntry of the entry state. 

So likely we would have to experiment a bit.

piegamesde, 2021-08-24:
> so, not having that boot state would kind of remove the onEntry of the entry state.

Yes. That would be the price, and in general this is okay because whatever logic was in `onEntry` could be moved to the reset state. The important thing is only to document it.

> So likely we would have to experiment a bit.

Any suggestions where I should start if I wanted to give this a try?

Dolu1990, 2021-08-25:
I just pushed a fix.

Now you can do : 

``` scala
  SimConfig.withFstWave.compile(new Component {
    val fsm = new StateMachine{
      val IDLE = bootAsEntry()
      val STATE_A, STATE_B, STATE_C = new State

      IDLE.whenIsActive(goto(STATE_A))
      STATE_A.whenIsActive(goto(STATE_B))
      STATE_B.whenIsActive(goto(STATE_C))
      STATE_C.whenIsActive(goto(STATE_B))
    }
  }).doSim{ dut =>
    dut.clockDomain.forkStimulus(10)
    dut.clockDomain.waitSampling(100)
  }
}
```

Basicaly, the IDLE state is the stateBoot of the statemachine with some makeup ^^
Seems to work well for me.

Dolu1990, 2021-08-25:
(if you have a better idea than bootAsEntry as a name, do not hesitate XD)

Dolu1990, 2021-08-25:
pimpMyBoot ?

piegamesde, 2021-08-25:
I'd go for a simple `bootState()` (or `getBootState()` if that's more idiomatic for Scala)

Dolu1990, 2021-08-25:
hmmm, i won't go for getBootState(), as that function change the behaviour of the statemachine. (bypass the regular entry state stuff)

What's about :

makeInstantEntry()

or

directBootState()  ?

piegamesde, 2021-08-25:
I like `makeInstantEntry` (or `makeInstantEntryState`), but also what about `modifyBootState` or `makeBootStateEntry`?

Also, this method is important to have some documentation, because what it does is really subtle and otherwise it is easy to miss it or to accidentally use it the wrong way.

Dolu1990, 2021-08-25:
Hmmm, makeInstantEntry is good for me :D

Sure, i will add the doc about it.

Dolu1990, 2021-08-25:
Pushed the makeInstantEntry fix.

Dolu1990, 2021-08-25:
documented : 
https://github.com/SpinalHDL/SpinalDoc-RTD/commit/492246e65ad69dc4f8ece0863e4d2814f15aa220

closed 2021-08-25 by Dolu1990