SpinalHDL/SpinalHDL #497: setDefinitionName is not escaped

piegamesde, 2021-08-24:
I can create a component with the name `Accumulator (128 bits)`. However, this will be used directly in Verilog without escaping, and thus produce syntax errors. Either restrict `setDefinitionName` to valid Verilog, or use escaping/normalization to produce a valid Verilog name.

Dolu1990, 2021-08-24:
Sorry, i don't understand, what do you mean by "this will be used directly in Verilog without escaping" ?

I don't understand the wording XD

Do you have an example ?

piegamesde, 2021-08-24:
Example: `setDefinitionName("foo bar (baz)")` generates `module foo bar (baz) (…)`

Dolu1990, 2021-08-25:
Hooo got it,

So farm the only protection which is made by SpinalHDL is to avoid duplicated names and reserved keywords.

So sure, this could be added

numero-744, 2022-11-22:
Hmm IMO if the user wants to have a name, this name should be used. Restricting to something like `[a-zA-Z_][0-9a-zA-Z_]*` could be done, but anyway if the name is trivially wrong, the simulator will report it during user tests; so that the user can change the name without surprises during code generation.

I would say it is better not implementing that (restricting the set of what the user can provide would be breaking change btw). Feel free to re-open this issue if you have counter arguments :wink:

closed 2022-11-22 by numero-744