SpinalHDL/SpinalHDL #498: Silent failure on wrong `switch` syntax

piegamesde, 2021-08-24:
I don't know how, but I happened to produce a `switch` statement with the following syntax:

```scala
switch(number) {
  0 -> {},
  1 -> {},
  2 -> {},
}
```

as per the documentation, this is not how the syntax of the switch statement goes. In simple cases, this silently generates valid Verilog. That Verilog code will then have broken, wrong or subtly different behavior than what the author intended. In more complex cases (for example writing a register in some way within an arm), this code triggers weird and seemingly unrelated compile errors like ASSIGNMENT OVERLAP.

Erroring out early on this wrong syntax would have saved me a lot of time while debugging this issue.

Dolu1990, 2021-08-24:
That is very weird, on which version of SpinalHDL / Scala are you ? 

if i try 

```scala
    switch(U(2)) {
      0 -> {},
      1 -> {},
      2 -> {},
    }
```

I get a 

';' expected but ',' found.
      0 -> {},

piegamesde, 2021-08-24:
Oh, I'm sorry, the "correct" wrong syntax is:

```scala
    switch(U(2)) {
      0 -> {}
      1 -> {}
      2 -> {}
    }
```

(no comma)

Dolu1990, 2021-08-25:
```scala
    switch(U(2)) {
      0 -> {}
      1 -> {}
      2 -> {}
    }
```

As far as i can see, it is impossible to catch that syntax, as  the only call to the SpinalHDL API is the switch(U(2)) 
All the x -> y stuff is some regular scala statement that do not interfer with SpinalHDL.
So, can't catch it any simple way. The only way i would see it catchable would be to extends the Scala compiler plugin implemented by SpinalHDL, but i would say it isn't a easy thing to do : 
https://github.com/SpinalHDL/SpinalHDL/blob/dev/idslplugin/src/main/scala/spinal/idslplugin/components/MainTransformer.scala

piegamesde, 2021-08-25:
I think this should be catchable at a type system level: If I read the code correctly, `switch` is a function that first wants a `BaseType` and then a `() => Unit`. On the other hand, the `x -> y` syntax generates a list (set?) of tuples that maybe get automatically collected to a map (not 100% sure, I'm not that familiar with Scala). This seems like incompatible types that should generate a type error at compile time.

Dolu1990, 2021-08-28:
> Maybe we can simply assert that applying block added at least one is or default statement and otherwise throw an error?

This would catch your case, but will also create issues with very generic code as : 

```scala
val elements = ArrayBuffer[(Int, () => Unit)]()
...
switch(cond){
  for((value, body) <- elements){
     is(value){ body() }
  }
}
```

In some way, it is a bit a similar consideration to zero width BitVector, it isn't necessarly a bug.

So, maybe throwin a warning,  and having a flag in the switch case to supress the warning ?

piegamesde, 2021-08-28:
To be honest, I think that `switches` should be exhaustive and I'm in favor of enforcing this in the compiler (disclaimer: I'm a Rust user :p ). Of course this is not feasible for Spinal but I don't think rejecting empty `switch` statements is wrong. You code snippet would work when written like this:

```scala
switch(cond){
  for((value, body) <- elements){
     is(value){ body() }
  }
  default { /* what happens on any other value */ }
}
```

Dolu1990, 2021-08-31:
> default { /* what happens on any other value */ }

The issue i have with it is that it often endup being a ceremony in the code instead of something usefull. In hardware i often try to reduce the area of the design, so, most of the case i have no default. 

> Of course this is not feasible for Spinal but I don't think rejecting empty switch statements is wrong.

From a staticaly human coded perspective, yes, i totaly agree. Issue is for programaticaly elaborated things.
So, i would be fine for a warning when a empty case appear, and a way to supress the warning, but else, i think it would be to many assumption on what are the usages of switch statements.

piegamesde, 2021-08-31:
> most of the case i have no default.

You wouldn't need one, as long as there is at least one guaranteed `is` statement.