SpinalHDL/SpinalHDL #499: Initialize TupleBundle with values

piegamesde, 2021-08-25:
May I present you a foot gun:

```
  val tuple = TupleBundle2(U(4, 32 bits), True)
```

Contrary to what one would expect, this code does **not** initialize `tuple` with `4` and `True`. The parameters are `HardType`, which ignores the values.

I think the correct solution would be to add an `object TupleBundle2` with `apply` methods that does the work. However, I'm not sure if this will work because of method overloading ambiguity.

Dolu1990, 2021-09-02:
So the usage that i would have for TupleBundle was for instance in the PLIC, where i had to do some data structure reduction / map / indexed accesses.

So, should we do the migration ? (i guess yess)

piegamesde, 2021-09-02:
I'm in favor of it too. However, the `HardType(TupleBundle2(Bool, UInt(8 bits))` use case should be documented with an example because it is highly non-obvious.

numero-744, 2022-11-22:
Do I perform the suggested change https://github.com/SpinalHDL/SpinalHDL/issues/499#issuecomment-906240706 in https://github.com/SpinalHDL/SpinalHDL/blob/dev/core/src/main/scala/spinal/core/TupleBundle.scala?

Btw I could add implicit conversion from Scala bundles to `TupleBundle`? IMHO tuples are interesting only because they are really short syntax to represent things, and have the drawback that their content is unnamed. Such an implicit conversion might perform conversions often, therefore creating many bundles; but if you use a bundle you don't care about (generated signal) names, right?

numero-744, 2022-11-22:
```scala
class TupleBundle2[
    T1 <: Data,
    T2 <: Data
    ](
        __1: T1,
        __2: T2
        ) extends Bundle {
  val _1 = __1
  val _2 = __2
}
```

I'm wondering about generated code:

1. `_1` is a new wire with a new name? If so, can it be assigned two-ways (does assignments works from `_1` to `__1` and `__1` to `_1`)?
2. Or is it just keeping the initial signal name when built from values, and signals are named `tuplename__1` and `tupelname__2` when creating a new one from hard types?

numero-744, 2022-11-22:
Also, maybe using an empty `case class` is better to avoid having different names in arguments and signal names?

numero-744, 2022-11-22:
What do you think about this? 

```scala
object TestTupleBundle extends App {
  // Could be overloaded for each TupleBundleN
  def T[T1 <: Data, T2 <: Data](_1: T1, _2: T2): TupleBundle2[T1, T2] =
    new TupleBundle2(_1, _2)

  // not taked into account for pattern matching so actually useless
  implicit def tupleBundle2ToT2[T1 <: Data, T2 <: Data](
      tb: TupleBundle2[T1, T2]
  ): (T1, T2) = (tb._1, tb._2)

  // This method could be added to expand signals
  implicit class tupleBundle2Pimper[T1 <: Data, T2 <: Data](
      tb: TupleBundle2[T1, T2]
  ) {
    def expand(): (T1, T2) = (tb._1, tb._2)
  }

  val a = T(True, False)

  // I wanted tupleBundle2toT2 to be used here
  //val (c, d) = a
  // But instead I can perform pattern matching thanks to case class
  val TupleBundle2(c, d) = a
  // Or with an added expand method
  val (e, f) = a.expand
  // In both cases, what about names?
}
```

Dolu1990, 2022-11-23:
> Btw I could add implicit conversion from Scala bundles to TupleBundle

Issue with it is that the returned TupleBundle elements will be typed as Data i think.

> _1 is a new wire with a new name?

It isn't

> Or is it just keeping the initial signal name when built from values, and signals are named tuplename__1 and tupelname__2 when creating a new one from hard types?

Yes

> an empty case class

?

So overall, the main question is : 

Independently of the user syntax. Should the TupleBundleX class create it's own instance of hardware signals, or should it reuse already existing signals and swap them in (as Vec(True, False, ...) does).

Currently, TupleBundleX create it's own signals, and the object TupleBundle allows to create a TupleBundleX with its  signals initialized with something. (which is kind of confusing from a user perspective XDXD)

So "if" the current TupleBundleX low level implementation is good, i would say let's keep it as it is, and add a cleaner raw userlevel API to be explicit about the intend (create new signal with or without init)

TupleBundle.init(True, False)  //Behave as the current TupleBundle(True, False)
TupleBundle.create(Bool(), Bool())  //Behave as the current TupleBundle2(Bool(), Bool())

Then we can think about easier / smoother user API.